roleoroleo/yi-hack-Allwinner-v2 #339: Minor upgrade for Outdoor is supported?

xorguy, 2022-01-20:
Hi.

My Yi Outdoor 1080p appears to have a minor firmware upgrade, 9.0.26.07_202104231123 and now have 9.0.26.05.

Because it is a minor upgrade I can upgrade and continue to use YiHack, isn’t it?
Best approach will be upgrade without the SD and then reinsert again? Or can be upgraded with the SD with YiHack?

Thanks.

roleoroleo, 2022-01-21:
I can not say for sure.
If you want, I can check it analyzing a backup.

xorguy, 2022-01-22:
Is it possible to upgrade and downgrade the firmware to try it? Or it will require play with partitions and so?

roleoroleo, 2022-01-22:
You have to prepare a partition file and copy it to the sd.
Is the same procedure used to unbrick the cam.
But you need the original partition.

closed 2023-01-08 by stale[bot]