roleoroleo/yi-hack-Allwinner-v2 #340: Hardcoded DNS at resolv.conf

xorguy, 2022-01-20:
Hi.

I just found out that file /temp/ resolv.conf have my DNS and then about 4 more servers, one being Google’s public DNS.

Are those DNS added by the firmware itself? In my opinion it is not ok because the device can use the other servers and skipping any kind of network control.

Is there any file where it can be changed? Because editing /tmp/resolv.conf it will be restored at next boot, right?

Thanks.

roleoroleo, 2022-01-21:
They are added by yi firmware.
You could remove it adding a script to the startup file.
After the boot, system.sh calls /tmp/sd/yi-hack/startup.sh
You can add your code to this file.

closed 2023-01-08 by stale[bot]

tempod, 2024-10-09:
I would also like to delete the dns that I am not interested in.
I created a file called startup.sh and put it in /tmp/sd/yi-hack/
inside this file I wrote
`sed -e '2,5d' /tmp/resolv.conf`
I restart the camera but when I go to check the file all the DNS are still present.
What am I doing wrong?

roleoroleo, 2024-10-11:
If I remember correctly DNS are handled by aYi process.
You should overwrite (maybe using mount bind) wifidhcp.sh and/or ethdhcp.sh.
These scripts are called when a network connection is established.
Check system.sh

darkfuncat, 2026-10-15:
I tried this fix but unfortunately it seems that the yi firmware update the file (once a day ?) without triggering wifidhcp.sh.
Instead I use wd.sh (watchdog ?) and the DNS are always under control.
Maybe it should be considered.