roleoroleo/yi-hack-Allwinner-v2 #342: Yi Dome U Pro 2K LFUS h60ga

pannal, 2022-01-24:
Hey there,

I've been using the Dome U Pro 2K with your hack and everything seems to work fine now.

The one thing that's baffling me is the abysmal quality of the stream - the RTSP stream and the recordings are fine, unless there's movement. With movement the moving object gets _extremely_ blocky and distorted to a point where it's not distinguishable from noise.

Is this normal? I thought the Dome U Pro 2K should at least have _somewhat_ acceptable quality.

Thank you!

pannal, 2022-01-24:
Probably a duplicate of #98

pannal, 2022-01-24:
I can confirm, this motion "garbage" only happens with the High stream. The low stream looks worse but doesn't have the heavy noise/distortion issues.

roleoroleo, 2022-01-26:
Sorry but this is the cam.
You coul try to disable all unnecessary services.
Maybe with more free resources it works better.

pannal, 2022-01-26:
Well, sadly there are no unnecessary services enabled; the only enabled ones are SSH and HTTPD. This camera is practically unusable this way.

Any known-good Yi camera you can recommend, that'll handle the basic functionality without issues?

Thanks!

roleoroleo, 2022-01-26:
> Edit: I have a YI Dome Guard (KFUSY36LA6ADGH200808) lying around.

Try with this cam.
My y21ga works well.

pannal, 2022-01-26:
Yeah the r30gb is _markedly_ better. Thanks! I'll return my Dome U Pro's.

closed 2023-01-08 by stale[bot]