roleoroleo/yi-hack-Allwinner-v2 #343: Yi Dome Guard r30gb RTSP/ONVIF dies after toggling privacy mode

pannal, 2022-01-26:
The Yi Dome Guard KFUS r30gb (9.0.05.32_202004101552) on 0.1.9 works way better than the h60ga I've tried before.

There's one pretty massive issue, though: After toggling privacy mode, RSTP and ONVIF completely die. 
The RTSPServer process is running, but neither VLC, nor ONVIF device manager, nor Home Assistant WebRTC can open any stream.

The only way to fix this is to reboot the camera.

Help!

pannal, 2022-01-26:
Killing the rRTSPServer process and waiting for `wd_rtsp.sh` to bring it back up again, works as well.

pannal, 2022-01-26:
Calling this after disabling the privacy mode "fixes" it as well: `http://yicam_ip:8080/cgi-bin/service.sh?name=rtsp&action=stop&param1=both`

pannal, 2022-01-26:
A couple of things:
* Is toggling privacy mode the right way, or should I use the camera_settings switch_on on/off service?
* Is the r35gb better in terms of hardware? Would that be [this one](aliexpress.com/item/1005002717138818.html?gatewayAdapt=glo2deu)? (My current one is [this one](https://smile.amazon.de/gp/product/B082D6NC14/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1), but I need two more)
* Do you have a recommendation for the current best PTZ YI indoor camera that's supported by your or other's hacks and buyable? It's really hard to get through all the different versions YI has.

ibanmda, 2022-01-27:
> The Yi Dome Guard KFUS r30gb (9.0.05.32_202004101552) on 0.1.9 works way better than the h60ga I've tried before.
> 
> There's one pretty massive issue, though: After toggling privacy mode (on then off), RTSP and ONVIF completely die. The RTSPServer process is running, but neither VLC, nor ONVIF device manager, nor Home Assistant WebRTC can open any stream.
> 
> The only way to fix this is to reboot the camera.
> 
> Help!

I have same issue with Yi Dome Guard (IFUSY), 0.2 yihack ver.

With muy others cams (Yi Home Camera) i don't have this issues!

roleoroleo, 2022-01-27:
>     * Is toggling privacy mode the right way, or should I use the camera_settings switch_on on/off service?

It depends on what is your goal.
The meaning of the privacy switch is to switch off rtsp and snapshot without shutting down the cam.
i.e. when I come back to home privacy switches on and when I leave home privacy switches off.
The switch "Switch on/off the camera" has the same meaning but only for yi apps
So, if it's switched off, rtsp and snapshot are on. And viceversa.

Finally, if you use service.sh web service, you can switch on/off all services you want:

- rtsp
- onvif
- wsdd
- ftpd
- mqtt
- mp4record
- all

But, at the moment, there is no snapshot in the list.

>     * Is the r35gb better in terms of hardware? Would that be [this one](aliexpress.com/item/1005002717138818.html?gatewayAdapt=glo2deu)? (My current one is [this one](https://smile.amazon.de/gp/product/B082D6NC14/ref=ppx_yo_dt_b_asin_title_o05_s00?ie=UTF8&psc=1), but I need two more)

The cpu is the same. I don't know, about sensor.
They are probably very similar.

>     * Do you have a recommendation for the current best PTZ YI indoor camera that's supported by your or other's hacks and buyable? It's really hard to get through all the different versions YI has.

No, maybe other users.

> Another "issue" I've found: After issuing PTZ commands while Privacy is on, then disabling Privacy again, the camera immediately detects motion and records a minute of footage. (Cloud off)

This behavior is strange.
I will check it.

About the issue in the title, I can't understand why.
I tested with my cam and it works.
But please, try this new file:
[privacy.sh.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7950509/privacy.sh.gz)

pannal, 2022-01-27:
> The cpu is the same. I don't know, about sensor.
They are probably very similar.

Thanks, I think I'll try and grab two of the r35gb as they've got a newer firmware and supposedly their PTZ presets work vs. the r30gb.

> About the issue in the title, I can't understand why.
I tested with my cam and it works.
But please, try this new file:

Applying, will report back, thank you!

pannal, 2022-01-27:
> But please, try this new file:
privacy.sh.gz

Awesome, this works! Thank you so much!

As a side note: Is there any reason to use the alternative RTSP server that you've added with 0.2.0? What's the difference?

roleoroleo, 2022-01-27:
> As a side note: Is there any reason to use the alternative RTSP server that you've added with 0.2.0? What's the difference?

Many users reported compatibility problems with NVR devices.
The rtsp stream occasionally stops working.
Check this issue: https://github.com/roleoroleo/yi-hack-MStar/issues/414

pannal, 2022-01-27:
> > As a side note: Is there any reason to use the alternative RTSP server that you've added with 0.2.0? What's the difference?
> 
> Many users reported compatibility problems with NVR devices. The rtsp stream occasionally stops working. Check this issue: [roleoroleo/yi-hack-MStar#414](https://github.com/roleoroleo/yi-hack-MStar/issues/414)

Hmm. The alternative RTSP server doesn't work for me. It's not running after a restart and toggling privacy doesn't run it, either. I think it might be crashing.

pannal, 2022-01-27:
One thing I've noticed while trying to debug this is that rtsp_server_yi doesn't have the same `-m MODEL` parameter as rRTSPServer has, but it's still being used by your scripts as if it would have.

roleoroleo, 2022-01-27:
`-m MODEL` parameter is ignored.
Simply, I'm using the same command line fot both daemons.
But you have a r30gb, so you must use `-C h265`

The last privacy.sh from this post
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/343#issuecomment-1023173879
shoudl work with rtsp_server_yi. Did you try?

> It's not running after a restart

Are you sure?
Could you share your system.sh?

pannal, 2022-01-27:
> `-m MODEL` parameter is ignored. Simply, I'm using the same command line fot both daemons. But you have a r30gb, so you must use `-C h265`

Huh. I didn't have to, before.

> 
> The last privacy.sh from this post [#343 (comment)](https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/343#issuecomment-1023173879) shoudl work with rtsp_server_yi. Did you try?

Yes, I'm using that one.

> > It's not running after a restart
> 
> Are you sure? Could you share your system.sh?

[system.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/7952307/system.zip)

roleoroleo, 2022-01-27:
> Shouldn't this be part of the r30gb 0.2.0 release, then?

rtsp_server_yi is experimental. I need to fix some bug in the next release.

> Yes, I'm using that one.

I'm missing something, I will check.

pannal, 2022-01-27:
> > Shouldn't this be part of the r30gb 0.2.0 release, then?
> 
> rtsp_server_yi is experimental. I need to fix some bug in the next release.
> 
> > Yes, I'm using that one.
> 
> I'm missing something, I will check.

FWIW: I updated from 0.1.9 to 0.2.0 using the Web UI. Maybe that wasn't 100% successful? (Everything else is working, though)

roleoroleo, 2022-01-27:
Your system.sh is ok, the upgrade worked properly.

1-Al, 2022-02-02:
@pannal sorry for being OT but does your camera produce it's own timestamp overlay?

alexdrl, 2022-02-23:
Hey, having the same issue on 0.2.0 and the se r30gb model, is that expected to be fixed in a following release or I am missing something?

roleoroleo, 2022-02-24:
It should be fixed in the next release.

pannal, 2022-03-12:
> > Another "issue" I've found: After issuing PTZ commands while Privacy is on, then disabling Privacy again, the camera immediately detects motion and records a minute of footage. (Cloud off)
> 
> This behavior is strange. I will check it.

Happens with r30gb (KFUSY36LA), 9.0.05.32_202004101552 - doesn't happen with r30gb (KFUSY36O5), 9.0.05.33_202009251856 (no way to update the first one to the newer firmware).

pannal, 2022-03-12:
@roleoroleo would it be possible to flash the first one with the dump I have from the other one?

roleoroleo, 2022-03-12:
I think it's possible.
If you share both backups, I can check which partitions are different.

pannal, 2022-03-12:
> I think it's possible. If you share both backups, I can check which partitions are different.

Here you go. _1 is the older one.
[backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8238066/backup.zip)

pannal, 2022-03-15:
@roleoroleo did you have any time to look into the backup zip I've posted, by chance? Thank you!

pannal, 2022-03-18:
> All partitions are different but you can "upgrade" the old cam using home and backup partitions only: mtdblock3 and mtdblock4.

Thanks! How do I do that? The unbrick procedure just seems to handle mtdblock4, not 3.

roleoroleo, 2022-03-18:
Check this topic:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/305

pannal, 2022-04-06:
> Check this topic: #305

I've copied mtdblock4.bin to the unbrick/h30ga folder and ran `./build.sh hacked`.

That resulted in:
```
Image Name:   xiaoyi-backup
Created:      Wed Apr  6 02:56:15 2022
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    1376256 Bytes = 1344.00 KiB = 1.31 MiB
Load Address: 00000000
Entry Point:  00000000
2432+0 records in
2432+0 records out
1245184 bytes (1.2 MB, 1.2 MiB) copied, 0.00456203 s, 273 MB/s
Successfully mounted h30ga/mtdblock4.bin on h30ga/mnt
### Updating h30ga/backup/init.sh
cp: cannot stat 'h30ga/backup/init.sh': No such file or directory
sed: can't read h30ga/backup/tmp_init.sh: No such file or directory
sed: can't read h30ga/backup/tmp_init.sh: No such file or directory
```

Which results in a 2kb `backup_h30ga.gz`.

Running the same with "factory" instead of "hacked" results in a "proper" >1MB `backup_h30ga.gz`.

Is this intentional/known?

pannal, 2022-04-06:
OK, having `backup_h30ga` on the SD card does nothing. Even after modifying `mtdblock4.bin` slightly (changing one of the FF bytes). What am I missing? Thanks!

roleoroleo, 2022-04-07:
I think that changing a byte in mtdblock4.bin is not enough.
Probably when the script decompresses the file, it "fixes" your change.
But, surely if you alternate the file created with "factory" with the one created with "hacked", it must go.

closed 2023-01-08 by stale[bot]