roleoroleo/yi-hack-Allwinner-v2 #344: Home 1080 problem

TheAlchemist92, 2022-01-28:
I've installed the firmware on my Yi 1080 home for onvif.
The camera is recognized by the NVR but the video isnt working.
If I try to use the web service or go to the IP address it doent work, but the original app work as always.
where did I go wrongwhere did I go wrong?

roleoroleo, 2022-01-28:
Are you sure the hack is properly installed?
Try to open a telnet or ssh session.

TheAlchemist92, 2022-01-28:
I cant access to the camera trough ip or net. It only work with app. I
found that the firmware is installed only because the camera was found by
the nvr trough ONVIF

roleoroleo, 2022-01-29:
If onvif works, telnet must work.
Try a telnet connection with putty.

TheAlchemist92, 2022-02-01:
I've tried but I cant write my pw in putty, when I try to write something
in the pw field it stay blank, I dont know why.
The login is the same of the app login, right?

roleoroleo, 2022-02-02:
> The login is the same of the app login, right?

Right.

TheAlchemist92, 2022-02-02:
Putty always says "login incorrect" if I use the app login data (email -
pw), I dont know why. Ive tried to ping the camera and it works.
My NVR views the camera but doesnt show me the image. If I change the pw
the NVR always shows the camera with a green dot, so I think that it
doesnt recognize the login data.

roleoroleo, 2022-02-11:
You don't have to use app credentials.
Default is root without password.

closed 2023-01-08 by stale[bot]