roleoroleo/yi-hack-Allwinner-v2 #345: Online FW upgrade from 0.1.9 to 0.2.0 does not get saved

querex007, 2022-01-29:
Online FW upgrade from 0.1.9 to 0.2.0 does not get saved
After click the upgrade firmware button behind the "Installed: 0.1.9 - Available: 0.2.0" message.
The firmware is still 0.1.9 after a successfull download / reboot.
Anything of debug info which I can check?
Please let me know.

Base version: 9.0.19.09_202010291747
Model suffix: y21ga
SD card intalled free / total: 36476/60912 KB

roleoroleo, 2022-01-29:
Any message in the upgrade text box?

querex007, 2022-01-29:
Message sequence in the text box seems ok:
Installed: 0.1.9 - Available: 0.2.0
Click on update button
Firmware download in progress
Download completed

querex007, 2022-01-30:
Anything else which I can test / consult some log files / update with SSH?

roleoroleo, 2022-02-01:
Maybe a problem with the sd card.
I don't know.
If the problem is not solved, please reinstall from scratch.

querex007, 2022-02-10:
Had simular issue on x2 1080p home models (BFUS and IFUS hardware).
IFUS does not want to upgrade OTA from 0.1.9 to 0.2.0, after reboot 0.1.9 was still installed.
BFUS did a "good" OAT upgrade from 0.1.9 to 0.2.0, but get into a reboot loop every +/- 25s after "succesfull" start, you can open the web GUI in the 25s every time again.

Both issues resolved when I did a fresh reinstall from scratch...

roleoroleo, 2022-02-11:
Thank for your feedback.
I will check the upgrade procedure.

closed 2023-01-08 by stale[bot]