roleoroleo/yi-hack-Allwinner-v2 #347: DHCP Client gets wrong IP

SeriousPat, 2022-01-31:
Hi,
i set up in my DHCP Server that my Oudoor 1080p Cam gets always the same IP x.x.x.115.
For some reason after soma days it has the ip x.x.x.201

Can i set a static ip?

Feature Request for ststic ip in Webinterface?

SeriousPat, 2022-02-03:
I have a reservation at my router. Thats why i am aksing. i don't know why the cam gets sometimes a wring ip.
Bit your link is a good solution. i tried it with the startup.sh its working.

in ssh: vi /tmp/sd/yi-hack/startup.sh

insert:
```
#!/bin/sh
killall -9 udhcpc
ifconfig wlan0 192.168.1.127 netmask 255.255.255.0 up
route add default gw 192.168.1.1
echo "nameserver 192.168.1.1" >> /tmp/resolv.conf
```

then in ssh: chmod 755 /tmp/sd/yi-hack/startup.sh and reboot

SeriousPat, 2022-02-04:
ok guys, sorry but the error occours again.
i use the startup.sh as show below.
yester evening i recognize the the cam has the ip 192.168.1.201
there are some logs?

SeriousPat, 2022-02-06:
Some news.
I disabled the setting in my router/dhcp that the cams get everytime the same ip.
after restarting the cams (the boot up with the startup.sh and static ip) everything is fine.
after 2 days, ine cam has an ip from my dhcp pool, the other cam again 192.168.1.201

roleoroleo, 2022-02-06:
Maybe the same mac addresses?
It shouldn't, but it can't happen.

SeriousPat, 2022-02-06:
Checked MAC, no, every mac is different

SeriousPat, 2022-02-06:
> Maybe the same mac addresses? It shouldn't, but it can't happen.

I did more tests. It does not matter if i reserve an ip in my dhcp server or not. after some hours/days, one of my 3 YI Oudoor 1080p IFUS Cam gets an ip that ends with .201
which cam it gets the ip is different, when one has allready 201, the next get 202.

closed 2023-01-08 by stale[bot]