roleoroleo/yi-hack-Allwinner-v2 #348: Wrong gain paramater in speak.sh and speaker.sh

VincentBerree, 2022-02-03:
I discovered the project some weeks ago and it's really great.
Lately I was fighting to get the TTS working in version 0.2.0 (for model y21ga): I was able to get sound in cli without pcmvol, but not via the WebUI. I check the code of speak.sh and notice the gain paramater "-g" for pcmvol whereas the WebUI set the volume in dB, so the gain parameter should be "-G" (as per pcmvol help).
I found the same issue in speaker.sh
After replacing -g by -G in these 2 files, all works fine.

MrTOffice, 2022-02-09:
Sorry to ask very basic question. How can I install yi-hack-utils from https://github.com/roleoroleo/yi-hack-utils? I am not familiar with linux.
Download *.tgz, *.tar.gz from the web and put it in which folder? Please guide. Thank you in advance.

alexfranke, 2022-02-10:
> and put it in which folder?

judging by the content of speak.sh, it looks like it's supposed to go in the sd card in yi-hack

 /tmp/sd/yi-hack/bin/nanotts 
 /tmp/sd/yi-hack/usr/share/pico/lang

I'm still trying to get it to work on y211ga, though... don't even see it in the UI. I can play a file using ssh by dropping it into the right place, but I've come up empty so far with the API, so I wonder if it's just not supported on this device.

alexfranke, 2022-02-10:
@MrTOffice  Okay I got it working, at least for y211ga.
- Those are the correct folders that I posted. 
- The web UI is accessible by specifying page=speak in the querystring 
- The comment from @VincentBerree is applicable. I went to /tmp/sd/yi-hack/www/cgi-bin and changed the lowercase g to an uppercase one after pcmvol, twice in speaker.sh, and once in speak.sh. That was a great find. 
- Works in both the UI and in postman. 

Now my cameras can threateningly chastise intruders in German.

roleoroleo, 2022-02-11:
Thank you for the feedback.
Sorry but I need to make a breaking change to fix this issue, in the next release, because I can't change the ha integration.
I added a new web cgi parameter "voldb".
If you want to change volume use "vol" for multiplier or "voldb" for db.
Please check these commits:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/36991d7ce9d6f773af094416c3b75293a958bccc
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/2aae35dbbaacf0c8462ce9b8d22b85997f143e55

VincentBerree, 2022-02-11:
The code looks good, but I can't figure out where I should put the file speak.js. Should it be concatenate in all.js then gziped under directory /tmp/sd/yi-hack/www/js?
I tested that, but it didn't work.

roleoroleo, 2022-02-11:
All js files are compressed to all.js
If you want to try it, this is the file:
[all.js.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8050501/all.js.gz)

VincentBerree, 2022-02-11:
Thanks for the file. I can hear text from tts but it looks like the volume is always the same whatever the value in DB.
I didn't find yet where is the issue. I found at least that the value is well sent to speak.sh, so I guess the issue is in this file but find where/why.

MrTOffice, 2022-02-12:
@alexfranke 
Thank you very very much for your help. Now I can make it work.
My device is 
> y21ga 
> BFUS
TTS is working right after I did what you suggest (and no need to reboot camera).
For playback sound, at first I think it is not work. Since I download sample sound from the internet 
> (https://drive.google.com/file/d/10HoDS9ct9d_3tUSGclUX66qZtVmSLXTx/view)
the length of the song is 2:17minutes.
I took around 1 min for camera to play.

MrTOffice, 2022-02-12:
Can I ask some more questions?
If I want to call service speak / speaker from Home assistant when camera track motion, what API and parameter should I used.
I found someone on the web url 

> http://yi.cam:80/cgi-bin/speak.sh (or speaker.sh) 

but I don't know the parameter

should I use shellcommand (curl) or rest_command
I ran to the code (without knowledge of .sh, I knew pascal but longtime ago).
How can I pass POST_DATA to the API?

If possible please give me some example call command, I would appreciated. Thank you in advance.

VincentBerree, 2022-02-12:
@mrtoffice Please have a look to [speak.js](https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/master/src/www/httpd/htdocs/js/modules/speak.js)

MrTOffice, 2022-02-13:
@VincentBerree , @alexfranke Thank you again.
At first, according to speaker.js (from @VincentBerree suggestion). I understand the url part, but the thing I don't know is "data" part. Thank to @alexfranke example. I now can make "speak" to work.

One more thing, I cannot pass file to make "speaker" to work.

`rest_command:
  camera_announce:
#    url: http://[camera address]:8080/cgi-bin/speak.sh?lang={{language}}&voldb={{volume}}
# I did no update speak.sh and speaker.sh, so I still use "vol" instead of "voldb"
    url: http://192.168.xx.xx/cgi-bin/speak.sh?lang={{language}}&vol={{volume}}
    method: POST
    payload: "{{message}}"

  camera_announce_wav:
    url: http://192.168.xx.xx/cgi-bin/speaker.sh?vol={{wav_volume}}
    method: POST
    payload: '{{wave_file}}'`

and service speak work just super fine (according to the example again). However in speaker part I tried to pass file to camera but nothing work. I know the problem part is how to pass append file to rest_command.

`service: rest_command.camera_announce_wav
data: 
  wav_volume: '-8'
  wave_file: '@/config/www/camera_sound/test.wav'`

`wave_file: '/config/www/camera_sound/test.wav'`
`wave_file: '@/config/local/camera_sound/test.wav'`
`wave_file: '/config/local/camera_sound/test.wav'`
`wave_file: '@/local/camera_sound/test.wav'`
`wave_file: '/local/camera_sound/test.wav'`
`wave_file: '@local/camera_sound/test.wav'`
`wave_file: 'local/camera_sound/test.wav'`

none of the above cannot make camera to play test.wav file.
Could you please guide me more?

roleoroleo, 2022-02-13:
> I can hear text from tts but it looks like the volume is always the same whatever the value in DB.

The volume does not change too much, the cam probably has an equalization system.

> Can I ask some more questions? If I want to call service speak / speaker from Home assistant when camera track motion, what API and parameter should I used. I found someone on the web url
>
>
> but I [...]

Check the wiki: https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Web-services-description
Or try my custom integration: https://github.com/roleoroleo/yi-hack_ha_integration

VincentBerree, 2022-02-14:
> > I can hear text from tts but it looks like the volume is always the same whatever the value in DB.
> 
> The volume does not change too much, the cam probably has an equalization system.

I guess the problem was between chair and keyboard. I uploaded again last version of speak.sh, speaker.sh and all.js.gz, and all works fine now.
Thanks.

closed 2022-02-14 by VincentBerree

MrTOffice, 2022-02-15:
Sorry to disturb you again.
Thank you very much for the wiki: **https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Web-services-description**
Also, I tried **custom integration** and it is very nice and very easy to config.

I myself really fascinated with the tts response time to process with the command and camera response. However, I am Thai (which pico don't have language for that). So, the thing I really need is "speaker" service (which the integration is not yet support, right now I find 2 services: ptz, speak)

I read speak.js and wiki but I still cannot "Add the raw audio in the payload" (Sorry about that). **Can you give me the example to  put "raw audio" in the payload?**

One more thing after run through speaker.sh (I guess it linux command which I am not familiar ), I think everytime to run speaker, it need to upload the wav file to the camera. Is there another way to reduce the time but put all wave files in camera and call service (like cat announce.wav > /tmp/audio_in_fifo but do it in rest_command)

PS: I am starting from Home Assistant (at first don't know about yaml too but it take a lot of time). So if possible please help. Thank you in advance.