roleoroleo/yi-hack-Allwinner-v2 #349: Private mode not working

norb-HB, 2022-02-03:
My Setup
YI 1080p Home Camera
Firmware Version | 0.2.0
Base Version | 9.0.19.02_202007141512
Model Suffix | y21ga

'Disable cloud'/'Private mode' seems not to work. Tested several combinations of setting in the hack but still able to see live pictures in YI app with LAN and LTE connect.

VincentBerree, 2022-02-07:
Have you reboot the cam after each change?
What is the content of /etc/hosts?
When 'Disable cloud'/'Private mode' option is enabled, the file should contain the following lines (after a reboot).
```
127.0.0.1    api.eu.xiaoyi.com
127.0.0.1    api.us.xiaoyi.com
127.0.0.1    api.xiaoyi.com.tw
127.0.0.1    log.eu.xiaoyi.com
127.0.0.1    log.us.xiaoyi.com
127.0.0.1    log.xiaoyi.com.tw
```

roleoroleo, 2022-02-11:
Did you solve the problem?

norb-HB, 2022-02-12:
Sorry for bothering you guys. It's working ok. The problem were sitting on fornt of the PC.

Thanks!

norb-HB, 2022-02-12:
Thanks! Private mode is working ok.

closed 2022-02-12 by norb-HB