roleoroleo/yi-hack-Allwinner-v2 #350: IFUS hard bricked

h4k3r2s1, 2022-02-04:
After power loss, camera died. Putty stops there:

=~=~=~=~=~=~=~=~=~=~=~= PuTTY log 2022.02.04 11:14:49 =~=~=~=~=~=~=~=~=~=~=~=
[233]HELLO! BOOT0 is starting Nov 14 2019 20:01:31!
[238]BOOT0 commit : 81c18ee
[241]board init start
[243]set pll start
[245]set pll end
[247][pmu]: bus read error
[250]board init ok
[251]chip id check OK
[254]DRAM BOOT DRIVE INFO: V0.41
[257]DRAM CLK = 528 MHz
[259]DRAM Type = 2 (2:DDR2,3:DDR3)
[262]DRAMC read ODT  off.
[265]DRAM ODT off.
[267]DRAM SIZE =64 M
[274]DRAM simple test OK.
[277]rtc standby flag is 0x0, super standby flag is 0x0
[282]dram size =64
[285]spinor id is: c8 40 17, read cmd: 03
[289]Succeed in reading toc file head.
[292]The size of toc is 4c000.
[345]Entry_name        = optee
[348]Entry_name        = u-boot
[352]Entry_name        = dtb
[356]Jump to secend Boot.
MESSAGE: [0x0] TEE-CORE: OP-TEE version: sun8iw19p1_v0.6.0-12-g97f2688 #1 2019年 11月 08日 星期五 08:26:51 UTC arm
ERROR:   [0x0] TEE-CORE:platform_standby_fdt_parse:126: no pmu node
ERROR:   [0x0] TEE-CORE:sunxi_twi_parse_from_dt:84: no pmu node

BTW for unbrick you can use micro usb cable instead of soldering just find old usb cable, cut it and you can use as UART:
black - GND
red - 5V
white - RxD
green - TxD
[image]

1-Al, 2022-02-04:
> BTW for unbrick you can use micro usb cable instead of soldering just find old usb cable, cut it and you can use as UART: black - GND red - 5V white - RxD green - TxD 

This depends on which camera, for h60ga it is possible [https://github.com/roleoroleo/yi-hack-Allwinner-v2/discussions/254](url)

Without u-boot you still have an option of SPI programmer as shown also in that link. There's also a possibility for recovery using an SDCard but hasn't been done yet AFAIK.

h4k3r2s1, 2022-02-04:
Someone have backup for y211ga?

alexfranke, 2022-02-04:
@h4k3r2s1 I have the one from the backup.tar.gz package -- is that what you're looking for? how do I get it to you?

h4k3r2s1, 2022-02-04:
I need all partition. If you have it sent me plz. I try unbrick my camera :D

alexfranke, 2022-02-04:
I'll leave it here for a few hrs https://drive.google.com/file/d/1lHdr8Qyg7qPse1D0cwZMPrhLQwk802GO/view?usp=sharing

closed 2023-01-08 by stale[bot]