roleoroleo/yi-hack-Allwinner-v2 #352: RTSP Low Quality doesnt Work

SeriousPat, 2022-02-08:
Hi,
on 2 of my three IFUS Outdoor 1080p the RTSP Low mode doesnt work.
My rtsp player (android) or VLC (Windows) connect, but doesnt show a picture. (black video)

roleoroleo, 2022-02-11:
Same versione and same hack?

SeriousPat, 2022-02-11:
Hi, yes, i use 3 cams, all the same version and this hack (allwinnerv2) 0.2.0

roleoroleo, 2022-02-11:
Is high res video ok?
Is low res snapshot ok?
Please take a screenshot of the statistics:
[image]

roleoroleo, 2022-02-26:
> upgraded all 3 cameras to 0.2.0.

Same model?

mspinolo, 2022-02-26:
Yes same model, 1080P BFUS

mspinolo, 2022-02-26:
Not easy unluckily: it is in a remote place

mspinolo, 2022-02-27:
so this is what I get

**1st camera**
```
root@CameraRose:~# cat /dev/mtdblock0 | grep uboot_
uboot_pkg magic 0x%x
uboot_ver
/uboot_y21ga
 uboot_message=%s(%s-%s)
```

**2nd camera**
```
root@CameraBarba:~# cat /dev/mtdblock0 | grep uboot_
uboot_pkg magic 0x%x
uboot_ver
/uboot_y21ga
 uboot_message=%s(%s-%s)

```
**3rd camera - one with issues**
```
root@YiCameraCorte:~# cat /dev/mtdblock0 | grep uboot_
uboot_pkg magic 0x%x
uboot_ver
/uboot_y21ga
 uboot_message=%s(%s-%s)
```

roleoroleo, 2022-02-27:
The cam are ok.
There is no reason why the cam is not working.
Try to change the sd card.

closed 2023-01-08 by stale[bot]