roleoroleo/yi-hack-Allwinner-v2 #356: enhancement in privacy.sh

VincentBerree, 2022-02-11:
I was playing with the privacy button in Home Assistant. Activating the privacy works fine by disabling RTSP.
But I notice once I deactivate the privacy, it starts mp4record whereas it was not started before. Would it be possible to check if mp4record is started before, store the status and restore the same when deactivating the privacy?
I was wondering why the camera status (ipc_cmd -t <on/off>), status led (ipc_cmd -l <on/off>) tracking and detections are not changed with privacy. Should we understand that once in privacy, the detections are still working?
If the detection is disabled with privacy, I suggest to disable the led to reassure people.

VincentBerree, 2022-02-12:
> Done, next release.
thanks :)
> Video stream and detections are not stopped, the privacy switch just stops rtsp server and snapshot.
Hopefully I could also stop mqttv4 to avoid sending the detection notifications.
If rstp server snapshot and mqtt (in a next release ;-) ) are stopped, why do not stop video stream and detections by forcing the status (like on/off switch)?
Is there a case that may need video stream and detections but no rstp nor mqtt?

roleoroleo, 2022-02-12:
> Is there a case that may need video stream and detections but no rstp nor mqtt?

I don't know. Users' imagination has no limits.
:)

It's not a problem to send the ipc_cmd and switch the on/off switch when the user enable/disable privacy.
The problem is the opposite.

VincentBerree, 2022-02-12:
> The problem is the opposite.

I don't understand.

roleoroleo, 2022-02-13:
I can't read the status of the on/off switch because I don't know where it's stored.
Probably somewhere in /dev/mtdblock7

VincentBerree, 2022-02-14:
> I can't read the status of the on/off switch because I don't know where it's stored. Probably somewhere in /dev/mtdblock7

Can't we force the status without reading it? I though this is how the switches work in camera settings page.

roleoroleo, 2022-02-15:
Yes, we could.

closed 2023-01-08 by stale[bot]