roleoroleo/yi-hack-Allwinner-v2 #363: Wifi disconnect from the router and App with error 3006

Svartskog, 2022-02-20:
Hi,

i own the Yi1080 Home QFUS which has been successfully flashed with the firmware
Firmware Version | 0.2.0
Base Version | 9.0.36.03_202109231441
Model Suffix | y211ga

Everythig works fine so far but arround every 6 Hour the camera loose the wifi connection and do not reconnect with my router.
Wifi strenght is arround 95%.

If i open the Yi App is see only the failure 3006, which only means : "The YI camera error 3006 means that the camera is offline. This usually happens when the server did not receive a heartbeat from the camera for an extended period, normally 5 minutes."

My wifi and inet connection is very stable...all my wifi devices like self flashes power sockets with tasmota works perfectly 24/7/365.
I try ever solution i could find ...static IP adress, turn on private mode but nothing helped.
I thought it was a faulty camera so I ordered two others but all have the same fault with the org firmware and with the hacked firmware.

But the cam seems to continue to run as it still switches between day and night mode, for example.(and record videos on SD card)
The only thing that helps is to unplug it and reconnect it to power.

does anyone have any ideas for this?

ykhandler, 2022-02-21:
> Hi,
> 
> i own the Yi1080 Home QFUS which has been successfully flashed with the firmware Firmware Version | 0.2.0 Base Version | 9.0.36.03_202109231441 Model Suffix | y211ga
> 
> Everythig works fine so far but arround every 6 Hour the camera loose the wifi connection and do not reconnect with my router. Wifi strenght is arround 95%.
> 
> If i open the Yi App I see only the failure 3006, which only means : "The YI camera error 3006 means that the camera is offline. This usually happens when the server did not receive a heartbeat from the camera for an extended period, normally 5 minutes."
> 
> My wifi and inet connection is very stable...all my wifi devices like self flashes power sockets with tasmota works perfectly 24/7/365. I try ever solution i could find ...static IP adress, turn on private mode but nothing helped. I thought it was a faulty camera so I ordered two others but all have the same fault with the org firmware and with the hacked firmware.
> 
> But the cam seems to continue to run as it still switches between day and night mode, for example.(and record videos on SD card) The only thing that helps is to unplug it and reconnect it to power.
> 
> does anyone have any ideas for this?

As i know, you just need to force close the Yi App than reopen it and try to view it again

Svartskog, 2022-02-25:
it has nothing to do with the app as the camera completely loses the connection to the router.

I also bought the 1080 V3 camera and it runs 24 hours without connection loss (org. Firmware)

closed 2023-01-08 by stale[bot]