roleoroleo/yi-hack-Allwinner-v2 #366: How to use speak volume and snapshot delay

PilaScat, 2022-02-21:
I saw that you updated and added the feature that I requested, thanks!
But how can I use them?

roleoroleo, 2022-02-21:
You can set snapshot delay in the web gui, mqtt section.
About speak volume, it's a web service feature. So you can use it only when called from another service.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Web-services-description#audio
If you use my ha custom integration, you will find a special option in the wizard.

PilaScat, 2022-02-21:
> You can set snapshot delay in the web gui, mqtt section. 
Thanks!
> About speak volume, it's a web service feature. So you can use it only when called from another service. https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Web-services-description#audio If you use my ha custom integration, you will find a special option in the wizard.
Can I change it without re-adding the integration? I have already changed all icons and names

roleoroleo, 2022-02-21:
I don't know.

closed 2023-01-08 by stale[bot]