roleoroleo/yi-hack-Allwinner-v2 #378: Beta release: 0.2.1_beta

roleoroleo, 2022-02-28:
Before to publish the new release, I would like to be sure that I didn't broke a couple of feature.
1- I completely changed the way how the grabber read the frames from the buffer. Both the grabber (h264grabber) and the rtsp server (rRTSPServer) are new.
2 - After several issues about the wrong watermark time, I moved the export of the TZ environment variable.

Anyone want to try the new version?
And check if:
1 - The rtsp stream is ok.
2 - The date/time is ok in the app, the date/time is ok in the web gui, the videos are playable in the app, the events page shows the proper date/times.

rpkilgore, 2022-03-01:
I'm attempting to test this beta version on y28ga, but having issues with it rebooting contantly, and the normal fix to disable the audio didn't work, then I disabled everything else except ssh and web and it still continues to reboot.  I'm going to run the remove hack and then try to re-apply and see if that corrects it,
Will keep you updated.

roleoroleo, 2022-03-01:
I tested it before tu publish and it seems working fine with my y21ga.
If the problem persists, try the new unbrick procedure:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick

rpkilgore, 2022-03-01:
well damn.. I just revived this camera from what i thought was bricked a few days ago using that unbrick process, and now I've managed to brick it again, but this time that unbrick process isn't working :(  Going to unplug it for a bit and try again after lunch.. lol maybe being powered off for a bit will change something lol **fingerscrossed**

roleoroleo, 2022-03-02:
Which model suffix?

rpkilgore, 2022-03-02:
y28ga  The unbrick worked the first time around, this time nothing happens.

Rushmed, 2022-03-04:
Hi rolee.

Thanks for this release.
Just installed [r35gb_0.2.0.tgz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/0.2.1_beta/r35gb_0.2.0.tgz) on my Dome Guard.
So far:
- Cam timezone, video recording and video playback in the app are ok
- Snapshot in web UI is also fine but still two timecodes are shown
[image]
- Time and date at /index.html are ok
- Events page semms to be ok

The test ran about 30 minutes so far, I`ll leave the cam online and will have an eye on it.

rpkilgore, 2022-03-06:
Ok, finally got a chance to test out the beta on my Dome Guard Cam r35gb.  Already had the hack applied so just updated by deleting everything off the sd card and extracting the new hack.  Made my custom changes to the conf files and booted up.  1 Reboot later and it was up and running smoothly.
Results:

- Snapshot has the correct time EST but there are 2 timestamps one on the bottom right and one on the top left
- if i change the url to say no to the watermark the timestamp in the top left remains
- RSTP stream works smoothly, proper time in the upper left hand corner
- date time stamps are correct everywhere EXCEPT for the folder structure name is incorrect
- I can however scroll through the timeline in the app and watch recorded events in the app!
[image]

roleoroleo, 2022-03-09:
Than you for yuor feedback.
A couple of notes:

> Snapshot in web UI is also fine but still two timecodes are shown

Yes, I can't remove the watermark added by yi (upper left).

> For me its confusing that v2.2.1_beta is named v2.2.0.

Yes, my mistake.

> date time stamps are correct everywhere EXCEPT for the folder structure name is incorrect

The directory structure created by the yi processes is UTC and if I try to change it, dates and times are not consistent.

outofsight, 2022-04-09:
I'll join here because I'm experiencing the issue of playback from SD card of previous events recorded by my y21ga.
I made the following settings about timezone:
- in the camera: "CET-1CEST,M3.5.0,M10.5.0/3"
- in the app: "UTC" (as of page recommendation)
Is that wrong?

When I try to playback, I'll see often the Kami Cloud request. If/when I can start watch a video I can't seek back where I want: the app go to live preview or to a wrong time in the past.

rpkilgore, 2022-04-13:
@outofsight  You want to set both the Camera and the Hack to your timezone.  UTC is what the application uses in the background.

outofsight, 2022-04-13:
> @outofsight You want to set both the Camera and the Hack to your timezone. UTC is what the application uses in the background.

Thanks, I made UTC setting in the app because I read this on the Hack Settings page:

> Timezone | Timezone. Get your timezone string from this page and copy here. If you want to use this feature, please set time to "UTC +0 Coordinated universal time" in the app.

roleoroleo, 2022-04-16:
I will fix this problem in the next release.

rpkilgore, 2022-04-17:
> I will fix this problem in the next release.

What fix are you going to apply?  I thought this was working already?

roleoroleo, 2022-04-17:
There are 2 changes related to the date/time available in the next release.
- I moved the TZ settings so only hack programs read it.
- I changed mqttv4 because motion_files messages was wrong.

The beta version tested in this thread contain only the 1st change.

outofsight, 2022-07-08:
In the last release should I put the correct TZ both in Yi app and in Hack?

roleoroleo, 2022-07-08:
Yes.

closed 2023-01-07 by stale[bot]