roleoroleo/yi-hack-Allwinner-v2 #379: Bricked BFUS (Y21ga)

uklatt, 2022-03-01:
Hello roloeoroleo,

I got another BFUS with 9.0.19 firmware today.
I was able to install version 0.1.9.
Then I tried to update to 0.2.0 (from webinterface).
After that the camera seems not to work anymore.
Then I used y21ga_unhack.tar.gz on empty card to revert the hack.
This seems to work.
Then I installed again y21ga_0.1.9.tgz but now the cam seems to be bricked...
Yellow light stays on and no "click" of the shitter anymore.

I have a backup of all partitions but when I try the new unbrick procedure, nothing happens.
Yollow light is still on.

What exact files I need to reflash the cam (with hopefully working bootloader)?

I have no serial connection at the moment.

Uwe

roleoroleo, 2022-03-02:
The "know" issue is the free space.
backup partition is working with too little free space and it is enough to write some files to corrupt the partition.
For example the hack procedure writes 1 file to the backup partition.
And the old unhack procedure writes 1 file (the same as above).
But the new unbrick procedure works at bootloader level, so it's able to restore the backup partition even if it's corrupted.

You have to copy to the sd only the file backup_y21ga as described in the readme:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick

And if you create the file with the command `./build.sh hacked`, the hack procedure doesn't need to write to the backup partition (so it's better if you want to use the hack again).

uklatt, 2022-03-02:
I tried this (backup_y21ga), but nothing happens (yellow LED is still on).
The I tried with additional backup files...

Hopefully the bootloader is still working.
Any other ideas?

Uwe

roleoroleo, 2022-03-02:
Try another sd card or try to reformat it.

uklatt, 2022-03-02:
This did the job!
With 32GB card it doesnt work.
With another 2GB card I was able to "repair" the cam ;-)

Thanks
Uwe

uklatt, 2022-03-02:
I was also able to reanimate my old cam 8-)))))

Uwe

closed 2022-03-02 by uklatt

MarcinJanCiesielski, 2022-04-16:
Uwe
Could you please share your working backup.  I have trouble (can't flash yi-hack and  connect to wifi) with the same camera model as you mentioned.

uklatt, 2022-04-17:
Hi [MarcinJanCiesielski](https://github.com/MarcinJanCiesielski),
don't you have a backup?
My problem was a not working SD card.
You should try differnt cards...

Uwe

MarcinJanCiesielski, 2022-04-18:
Hi @uklatt 
I got YI Home 1080p cameras in such state (unable to connect to WiFi) - I did backup but it is the backup of such faulty state.
I want to repair them. At first I flash yi-hack (Flash folder was renamed to Flash.done) but nothing changed when it comes to connecting. Cameras are BFUS and the firmware version is 9.0.19.02_202007141512. My WiFi is 2.4 and is also OK - I have another YI camera (of different type) where I successfully flashed another yi-hack and connected to my WiFi.

My last hope is to flash original working backup from another cam and try to connect to WiFi.

Marcin

roleoroleo, 2022-04-18:
If the other cam is the same model (9.0.19.....) you can use it to restore the 1st cam.
Use only backup partition.

MarcinJanCiesielski, 2022-04-18:
@roleoroleo 
Thank you for answer.
I need clarify situation: I have 3 YI Home 1080p (same type: BFUS, same firmware 9.0.19.02_202007141512) but every single one has the same problem - unable to connect to WiFi. I have backups of cams, but if I swap backups the problem stays.
The came which works is YI Dome 1080p PTZ - it's completely different model and firmware so I can't use it.

roleoroleo, 2022-04-18:
Did you try this?
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials

MarcinJanCiesielski, 2022-04-19:
@roleoroleo 
Yes, I tried it. After this operation cams stayed disconnected, but I have no opportunity to check whether yi-hack is applied or not - I can only see that during flashing folder Flash is renamed to Flash.done and AFAIC having yi-hack applied is necessary to conduct operation you mentioned.
I read about similar problem (not connecting to WiFi) on 4pda (Yi Home 720p) - not enough space on partition so I thought my problem is similar and if I flash original blank partitions I will solve it.

roleoroleo, 2022-04-23:
You could try to flash the partition obtained with "hacked" argument.
If the problem is the free space, you should solve it.

MarcinJanCiesielski, 2022-04-26:
I downloaded "blank original" backup partition, prepared "unbrick file" according to new unbrick procedure. I flashed prepared file and I fixed all 3 cameras. All connect to WiFi flawlessly.

Thanks for interesting in my issue.