roleoroleo/yi-hack-Allwinner-v2 #380: Access media files from home assistant

oat1, 2022-03-02:
Hi is there a way to access these media files from home assistant ?

It would be nice to add this folder to media section and watch them directly from this section.

[image]

roleoroleo, 2022-03-02:
You can upload the file to an FTP server or access them with the web services.
Check the wiki:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Web-services-description

oat1, 2022-03-02:
Yes, but now finding a file is a pain in the ass.

I'm thinking about something more "user friendly"
Accessing camera sd card like a network drive, and put the video catalog to media section in HA
The whole main catalog of video files, let's say " Yihack", then subfolder with specific days, hours and inside a list of files to play. Just like frigate does ( screen below)
It would be much easier to watch and find a specific video from specific

I do not know whether it is possible or not, It is just a wish to consider :)
But I think it would be a great thing. 

[image]

roleoroleo, 2022-03-06:
Try the new release.

oat1, 2022-03-06:
Wow Great 😊👍 thanks . 
There is now a new folder yihackin media section but when I go inside, error appears

[image]
[image]

roleoroleo, 2022-03-09:
Please, check the registry log and send me the error.

This is what it should appear:

[image]

[image]

[image]

[image]

oat1, 2022-03-09:
Log below

Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/yi_hack/media_source.py:76
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 06:23:53 (3 occurrences)
Last logged: 08:39:47

[547642575360] Error handling message: Unknown error
[547642851184] Error handling message: Unknown error
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/decorators.py", line 27, in _handle_async_response
    await func(hass, connection, msg)
  File "/usr/src/homeassistant/homeassistant/components/media_source/__init__.py", line 152, in websocket_browse_media
    media = await async_browse_media(hass, msg.get("media_content_id", ""))
  File "/usr/src/homeassistant/homeassistant/components/media_source/__init__.py", line 111, in async_browse_media
    item = await _get_media_item(hass, media_content_id).async_browse()
  File "/usr/src/homeassistant/homeassistant/components/media_source/models.py", line 85, in async_browse
    return await self.async_media_source().async_browse_media(self)
  File "/config/custom_components/yi_hack/media_source.py", line 76, in async_browse_media
    domain = list(list(device.identifiers)[0])[0]
IndexError: list index out of range

roleoroleo, 2022-03-09:
Try this file:
[media_source.py.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8213566/media_source.py.gz)

oat1, 2022-03-09:
Now Works flawlessly,  Thanks! For the feature and for help.
You're the best !!!

[image]

roleoroleo, 2022-03-09:
With the next release cam/integration the video will have a thumbnail that will display the first frame.

oat1, 2022-03-09:
Great 👍😃 Couldn't ask for more. Appreciate the job

oat1, 2022-03-18:
Hi, I updated camera to the latest 0.2.1 release, but there is no thumbnail, still just icons
yi-hack Home Assistant integration is also updated to 0.3.8

[image]

roleoroleo, 2022-03-18:
Login with ssh and check if crontab file is ok:
```
root@yi-hack-allwinner:~# cat /var/spool/cron/crontabs/root
* * * * * /home/yi-hack/script/thumb.sh cron
0 * * * * sleep 20; /home/yi-hack/script/clean_records.sh 30
```

Try to run the process manually and check the output:
```
root@yi-hack-allwinner:~# . /home/yi-hack/script/env.sh
root@yi-hack-allwinner:~# /home/yi-hack/script/thumb.sh cron
```

oat1, 2022-03-18:
I get this

[image]

oat1, 2022-03-18:
I've edited it a little and process started 
[image]

roleoroleo, 2022-03-18:
Sorry, I got confused with Allwinner.
This is for Allwinner-v2

```
root@yi-hack-allwinner:~# cat /var/spool/cron/crontabs/root
* * * * * /tmp/sd/yi-hack/script/thumb.sh cron
0 * * * * sleep 20; /tmp/sd/yi-hack/script/clean_records.sh 30
```

Try to run the process manually and check the output:
```
root@yi-hack-allwinner:~# . /tmp/sd/yi-hack/script/env.sh
root@yi-hack-allwinner:~# /tmp/sd/yi-hack/script/thumb.sh cron
```

oat1, 2022-03-18:
It is running now, starting thumb.sh cron helped. 
Thumbnails are being created and I can see them in the media section.

To be honest I have no idea why the script did not run automatically after the update and reboot

oat1, 2022-03-18:
Proces makes thumbnails, but only once. New records do not have them and cron does not start automatically again. How do I change it ?

[image]

roleoroleo, 2022-03-18:
Is crond running?

oat1, 2022-03-18:
How do I check that?

roleoroleo, 2022-03-18:
```
root@yi-hack-allwinner-v2:~# ps | grep crond
```

oat1, 2022-03-18:
[image]

roleoroleo, 2022-03-18:
The process is ok, I don't understand.

> and I have
> 0 * * * * sleep 20; /tmp/sd/yi-hack/script/clean_records.sh 5

This is another process, don't care.
Do you have another line with thumb.sh or just this line with clean_records.sh?

oat1, 2022-03-18:
I have this lines

[image]

roleoroleo, 2022-03-18:
It's ok.
env.sh command sets environment variable.

Check if there are empty jpg in the record folder.
If the process find a jpg file it doesn't try to create it again.
But if the file is zero lenght you can't see the image in the media source.

oat1, 2022-03-18:
I did not find any empty jpg :(

If it is a cron then how often should the process start? Every 1 hr ?

roleoroleo, 2022-03-18:
"* * * * *" means every 1 minute.

roleoroleo, 2022-03-20:
Found the problem.
Check the last commit:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/968dd8580b1800c2b0ae0d035a171be490b21781

oat1, 2022-03-20:
Great 😃👍 can I test it now, upadte via ssh ? How?
Or wait till next release?

roleoroleo, 2022-03-20:
> upadte via ssh ?

Yes.

oat1, 2022-03-20:
But how ?

roleoroleo, 2022-03-20:
You can use vi to edit the file.
Or, download the file from github and copy it to the cam with ftp.

oat1, 2022-03-21:
I've applied these changes in ftppush.sh and thumb.sh, reboot the camera and the problem still appears. No new thumbnails are being created

oat1, 2022-03-21:
Sorry, after some time works just fine :) thanks for your patience :)  
My last question due to this topic, Can I somehow enable the thumbnail in [yi-hack-v5](https://github.com/alienatedsec/yi-hack-v5)?
I know that it is not your project, but you help the creator. Maybe I will copy some files and lines of code from this project to make it work?

closed 2023-01-08 by stale[bot]