roleoroleo/yi-hack-Allwinner-v2 #385: IR LEDs to bright

uklatt, 2022-03-05:
Hello,

I use the Yi camera in a nest box.
At night the IR leds are to bright.
Is there the possibility to reduce the brightness (perhaps with PWM control)?

If this is not possible, is there an option to reduce brightness with a filter?
What material would work?

Uwe

askanhesse, 2022-03-09:
Hi Uwe,

Just saw this issue coincidentally.
I got into the same issue when building my nest box with this camera some days ago.

I also took a look into this topic and it seems its not that easy to get this functionality yet with this kind of camera hacks.
Take a look on https://github.com/roleoroleo/yi-hack-Allwinner/issues/284#issuecomment-1013985697 for more context.

As a working solution even when you losing a bit of quality I can recommend to just put some electrical take to three of the six IR leds (e.g. two left, one right) on the inside of the camera. You just need to remove 4 screws from the outside cover and pull out the main camera module. After that you can remove the two screws of the IR module.
To get rid of the quite loud clicking of the camera when switching between day (color) and night (black+white) mode, you can force the camera to stay in the day mode by removing the connection to the mechanism in the lens, see https://github.com/roleoroleo/yi-hack-MStar/issues/347#issuecomment-997230777

Hope that helps!

Askan

uklatt, 2022-03-09:
Hi Askan,

thank you for your comment.
Yes, I should do this when there is no option to reduce the brightness with software.

Uwe

uklatt, 2022-03-10:
Hello,

perhaps this code can control the IR shutter:

```
    int driver = open("/dev/cpld_periph", O_RDWR);
    // IR filter OFF
    ioctl(driver, _IOC(_IOC_NONE, 0x70, 0x15, 0), 0);
    // IR filter ON
    ioctl(driver, _IOC(_IOC_NONE, 0x70, 0x16, 0), 0);
    close(driver);
```
I got this code from https://bestofcpp.com/repo/cjj25-Yi-RTS3903N-RTSPServer-cpp-video
There is also code for controlling the switching level. Perhaps this works also.

Uwe

askanhesse, 2022-03-10:
> What happens when only the shutter is disabled?
> Is the image with IR LEDs on still to bright or does the filter eliminates some IR light?

With all IR Leds still on? I would say its too bright, since I first tried with only two LED and for me it felt a bit too dark.

uklatt, 2022-03-10:
I mean, when the shutter is closed and IR LEDs are on.
Uwe

pannal, 2022-03-12:
> Hello,
> 
> perhaps this code can control the IR shutter:
> 
> ```
>     int driver = open("/dev/cpld_periph", O_RDWR);
>     // IR filter OFF
>     ioctl(driver, _IOC(_IOC_NONE, 0x70, 0x15, 0), 0);
>     // IR filter ON
>     ioctl(driver, _IOC(_IOC_NONE, 0x70, 0x16, 0), 0);
>     close(driver);
> ```
> 
> I got this code from https://github.com/cjj25/Yi-RTS3903N-RTSPServer There is also code for controlling the switching level. Perhaps this works also. (The code is for another camera type with RTS3903N.)
> 
> Uwe

This would be great to add for privacy mode @roleoroleo!
Even though my cameras are in privacy mode, they are still swapping between day/night mode depending on whether lights are on or off.

roleoroleo, 2022-03-12:
I can switch on and off the shutter but I can't prevent the cam from switching it.

uklatt, 2022-03-12:
Does it help to change switch levels?

Uwe

roleoroleo, 2022-03-12:
What do you mean?

uklatt, 2022-03-12:
The switching levels for the shutter (ADC Values) of the light intensity (image) should be adjustable.
Uwe

closed 2023-01-08 by stale[bot]