roleoroleo/yi-hack-Allwinner-v2 #388: y211ga Cameras. Was it a bad idea to update to 11.0.36.7_202202170022? Motion detection is completely silent.

dreimer1986, 2022-03-06:
I got two cams of the type y211ga. Before I installed the hack 0.2.0 I read through the discussions and as suggested there updated the cams with the official app first. Now I am at 11.0.36.7_202202170022 on both and have one problem. Motion detection is dead silent. I tried the official addin and MQTT and both say nothing when I hop around in my room. Detection is active in the official app, no paid service was chosen.

I made shots of the whole config. Maybe I am just too stupid: https://photos.app.goo.gl/Eur5NTmZDA2zyu1y7

Any idea?

dreimer1986, 2022-03-08:
[backup QFUSY65V64.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8204593/backup.QFUSY65V64.zip)

Full Backup script made dump off the new firmware. Maybe useful? This one was made BEFORE hacking.

dreimer1986, 2022-03-09:
A friend bought the same cameras and yet did not update or hack anything. This is the firmware in v9 range
[backup_v9.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8212718/backup_v9.zip)

roleoroleo, 2022-03-09:
About the comparison between the 2 versions above, it's not clear.
For sure home and backup partitions are upgraded (mtdblock3 and mtdblock4 are different).
But the other partitions are also different, and I think that nobody would do the OTA update of the bootloader, kernel, etc...
I don't know...

HolgerOnGithub, 2022-03-14:
Is there a possibility of downgrading the firmware? I also have the 11.0.36.7_202202170022 and no motion detection after the hack. Of course I only have a backup with version 11.0.36.7_202202170022, but here is a zip with version 9.0.36.02_202105081647. How do I get it into my YI 1080p Home IFUS?
Thanks in advance

dreimer1986, 2022-03-17:
There is the way the former unbrick hard way says... Soldering and Serial on UBoot. I have no clue if it's doable to use DD command via telnet to overwrite the partitions. Likely it will die while doing that. RAW Overwriting the OS from the OS sounds like a terrible idea... I think if you don't want to solder (I would prefer not to) we have to wait until it's supported. That's why I passed over my 11.x backup and the 9.x one from my friend.

roleoroleo, 2022-03-17:
The soft unbrick is possible without soldering.
Check the wiki and this thread:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/305

dreimer1986, 2022-03-18:
Sure, but we have more than just partition4 aka Backup changed...

roleoroleo, 2022-03-18:
You can use the soft method for all partition.
But avoid to use it for u-boot.

dreimer1986, 2022-03-18:
Hmm, OK. Then I need to tinker a bit with that on weekend. I just read of the backup dump there, so I thought that's it. So it does use all dumps I give to it on recovery file creation?

roleoroleo, 2022-03-18:
You have to copy these files on your SD and reboot the cam (and wait 10 minutes).
Start with backup, then home.
Probably kernel and rootfs are not necessary.
Avoid env and uboot.

dreimer1986, 2022-03-18:
Alright, so one file only and then two 10 minute sessions for the two partitions. I will keep you updated. ^^

dreimer1986, 2022-03-18:
So, good news. I did as you said, first backup, then home AND skipped the kernel image. Result:

[image]

The app now detects movements and sometimes even sounds, without the Kami Cloud paid for.  But the sensors MQTT and official addin do not do much yet... Likely my MQTT settings are wrong? https://photos.app.goo.gl/Eur5NTmZDA2zyu1y7 Pics off the old non downgraded one and 0.2.0 Hack, but the settings I took 1:1 of these again. Do you see any stupidity here? Or is the kernel maybe still the problem here?

P.S: The 0.2.1 hack reduced the lag from RTSP down to less of a second :D

dreimer1986, 2022-03-18:
OK, I disabled all the advanced advertisement and now... IT WORKS !!!!! No clue why, but it works great :D 
OK, so I have to decide between motion and human detection, right?

Btw, the 11.X firmware claims to add animal, vehicle and improved person detection. (And Activity Zone for these and timestamps for the videos) Just to let you know that maybe the new firmware might be interesting to add somewhere in far future.

roleoroleo, 2022-03-18:
> OK, so I have to decide between motion and human detection, right?

Yes.

> Btw, the 11.X firmware claims to add animal, vehicle and improved person detection. (And Activity Zone for these and timestamps for the videos) Just to let you know that maybe the new firmware might be interesting to add somewhere in far future.

I'm not sure but I think that new feature are "server side".

dreimer1986, 2022-03-18:
Alright. In this case. First: THANKS for help! All is fine now and this way to downgrade can be used to get back to a working state for others. I will close this one now.

closed 2022-03-18 by dreimer1986

Minims, 2022-08-08:
> Alright. In this case. First: THANKS for help! All is fine now and this way to downgrade can be used to get back to a working state for others. I will close this one now.

@dreimer1986 How did you success a Revert to 9.0.36 ? I am trying to use the unbrick procedure, but nothing append, i am still in 11.xx version

dreimer1986, 2022-08-08:
Of course you are. You need a v9 backup of the same camera type and flash the partitions off it back on yours. That's what I did

Minims, 2022-08-08:
@dreimer1986 I have the backup v9 on my own camera.
I have build the files via the unblock Readme.
I have : 
```
backup_y211ga.gz
home_y211ga.gz
rootfs_y211ga.gz
```

Then I have : 
```
gunzip backup_y211ga.gz
gunzip home_y211ga.gz
gunzip rootfs_y211ga.gz
```

Format SDCARD to FAT32
Paste backup_y211ga, and the start camera.
Led blinks one time in yellow and then it start normally after 15sec. (Longer than normal boot)

Once done, I make the same with home & rootfs. But I am still in v11. I tried with the backup file on this thread, but same behavior. I might miss something :-/

dreimer1986, 2022-08-08:
I used the WIndows tool found there a bit hidden (https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/305#issuecomment-983783470) for making the files and I even skipped rootfs. Only backup and home on my cams was reverted. All fine here. And ALWAYS do ONE AT A TIME. Not all of them thrown on SD and boom!. DO one, remove it, do next.

meluvalli, 2022-08-09:
So to confirm, you have NOTHING on the flash drive except the 1 file?  Example: backup_y211ga?  How does the camera know to pick up the file and process it?

Minims, 2022-08-09:
I have finally be able to revert.

1. I have format SDCARD via FAT32 with Pi Imager. (If I format with the Yi Home App, it fails)
2. I need to use the file files from the backup_v9, use backup and then home file. (revert works)
3. I made the same with my own backup file, I have a newer 9.0.36 than the one on backup_v9
4. I reapplied the hack.
5. All is OK now.
It was a long journey

meluvalli, 2022-08-09:
@Minims 
Hmm...  I must be doing something wrong :(   

I downloaded the v9 zip file you posted in https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/388#issuecomment-1062645815.  
I downloaded the mkimg file from https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/305#issuecomment-987908753
I then ran the command in windows: 
```
mkimg.EXE -A arm -O linux -T filesystem -C none -a 0x0 -e 0x0 -n "xiaoyi-backup" -d mtdblock4.bin backup_y211ga
```
I used Pi Imager as you did just incase to Format the SD Card FAT32.
I then copied the backup_y211ga to the root of the SD card.  This is the ONLY file on the SD camera.
I inserted it into the camera, plugged it in and waited 10 minutes!  (Yes, I set a timer)... 
Then unplugged it.
I then copied the home_y211ga file and deleted the the backup_y211ga file to the SD card.  Again, only left with ONE file on SD Card called home_y211ga
I inserted it into the camera, then booted it.  Waited 10 minutes!  (Yes, I set a timer again!)... 
Then unplugged it.
Removed the SD Card, plugged the camera in.

YI Home still says I'm on version 11.  I don't think it did ANYTHING!  Am I missing something?

Minims, 2022-08-09:
@meluvalli does the yellow led has blink one time just after the boot ? 
 - I have remove the camera from Yi Home App
 - I have clean the application cache too (try this every time you update a file backup, home or rootfs)
 - Then I have rest the camera via reset button and had it again.

meluvalli, 2022-08-11:
Got it.   I think reset was the key :)

Thank you!