roleoroleo/yi-hack-Allwinner-v2 #392: Camera is always recording/saving/uploading to FTP.

Hr-, 2022-03-13:
Hi, great work with the hack!

My issue is the camera is always recording when motion is detection and saving files to sdcard (flooding my FTP server with video.) 

If "Save video when a motion is detected" enabled of disable it still make video files.
If "Motion Tracking sensor" is enabled of disable it still make video files.
If Home/Away is enable in app it enabled of disable it still make video files.

Is there any way to disable video saving when it app is "Home" so it doesn't save files or upload to FTP or is the app not "talking" to camera? 
Or is that just the way camera is made so is always recording and setting are set in app?

Tested with 
Firmware Version 0.2.0 and 0.2.1_beta
Base Version  9.0.36.02_202105081647
Yi 1080p Home IFUS | 9.0.36* | y211ga

Best regards
Hr

[config.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8239387/config.zip)

roleoroleo, 2022-06-07:
I don't understand the problem.
What is your desired behavior?
The cam should save a file only when a motion is detected.
The cam shouldn't save files at all.

Hr-, 2022-06-07:
@roleoroleo 

Don't know if it possible with how the app is talking to the cam.
But I would preferer that it only record and save video if "Away" is enabled in app.

Atm it's recording all detect and save regardless if you enable/disable the settings on cam.
"Save video when a motion is detected"
"Motion Tracking sensor"

roleoroleo, 2022-06-07:
> But I would preferer that it only record and save video if "Away" is enabled in app.

Sorry, it's not possible.

closed 2023-01-07 by stale[bot]