roleoroleo/yi-hack-Allwinner-v2 #393: Unbrick r30gb

ibanmda, 2022-03-13:
Hi  @roleoroleo

I bricked my Yi Dome Guard Camera with my backup after i didn't soft reset succesfully. The cam has permanent red light.

I've readed your how to hard unbrick, but how you can see in the photo my board hasn't GND, RX & TX to solder with my USB-TTL.

I'll aprecciate any suggestion.

Thanks!

[image]

ibanmda, 2022-03-13:
Maybe is this?

TP3 -> GND
TP2 -> TXD
TP1 -> RXD

ibanmda, 2022-03-23:
Ok! I tried again with soft method with my old backup and i had configured the cam with the YI app succesfully.
But when I try to hack with yi-hack after the reboot the led still permanently blink green without connection.

Any idea?

roleoroleo, 2022-03-25:
Try to restore again backup partition.
But load a backup file obtained with this procedure:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick
using the "hacked" option
`./build.sh hacked`

When you restart the cam, it should be already preapred fot the hack.
And the hack procedure shouln't brick the cam.

ibanmda, 2022-03-25:
I got install again the hack with 0.2.1 firm. I think that the mistake was in the wifi config. Once I copied the key and SSID the cam works perfect with the firmware. Therefore this post can be closed succesfully! :)

Thank you again @roleoroleo !!!

closed 2022-03-25 by ibanmda