roleoroleo/yi-hack-Allwinner-v2 #396: Segfault in rRTSPServer v0.2.1

foxel, 2022-03-18:
I'm using https://github.com/AlexxIT/WebRTC to play camera streams in HA.
Recent updated rRTSPServer is failing once the RTSPtoWebRTC connects to it:

```shell
root@yi-home-fourth:/tmp/sd/yi-hack# ./bin/rRTSPServer -m y21ga -r low -a ulaw -p 554 -u admin -w Password
1647579471585: warning - 298703 frame(s) lost

"ch0_1.h264" stream, from memory
PCM audio enabled
Play this stream using the URL "rtsp://192.168.0.164/ch0_1.h264"

"ch0_2.h264" stream, from memory
PCM audio enabled
Play this stream using the URL "rtsp://192.168.0.164/ch0_2.h264"
Segmentation fault (core dumped)
```

I've replaced the rRTSPServer file with the one from 0.2.0 and it works OK.

yi-hack-Allwinner-v2 version: **0.2.1**
WebRTC version: **v2.2.0**
YI Camera: **y21ga**

roleoroleo, 2022-03-18:
Only with WebRTC?
Does vlc work?
If you remove authentication?
I tested it now and I can't replicate the problem.

foxel, 2022-03-19:
VLC works fine.
I have a password setup.

Checked without password - works OK, even having the WebRTC set to send a password. Looks like the issue is somewhere in the authentication flow.

roleoroleo, 2022-03-19:
Try this:
I upgraded to [live.2022.02.07.tar.gz](http://www.live555.com/liveMedia/public/live.2022.02.07.tar.gz)

[rRTSPServer.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8308861/rRTSPServer.gz)

foxel, 2022-03-19:
works fine

closed 2023-01-08 by stale[bot]