roleoroleo/yi-hack-Allwinner-v2 #397: r30gb bootloop when enabling mqtt

lcorsini, 2022-03-18:
I have a strange problem with an r30gb
it was hacked before, but microsd died
got a new one, downloaded vers 0.21, it works fine until I enable MQTT config
any ideas? the only thing left to try is to reinstall 0.19 or keep mqtt disabled

I have another r30gb that works fine with mqtt I also upgraded it to 0.21 without problems

roleoroleo, 2022-03-19:
Is swap file enabled?

lcorsini, 2022-03-21:
no swap is enabled at the moment, no

roleoroleo, 2022-03-21:
Try to enable it.
Probably the problem is the snapshot.
When a motion is detected, the cam take a snapshot.
Snapshot requires a lot of memory.

lcorsini, 2022-03-21:
I'll try (also probably disabling cloud will help) there is a documented  procedure to enable swapfile? I've found this issue https://github.com/roleoroleo/yi-hack-Allwinner/issues/5

roleoroleo, 2022-03-21:
If you are able to reach the web page switch on the flag.
Otherwise, edit the file yi-hack/etc/system.conf in the sd card.

lcorsini, 2022-03-22:
it seems to work now, so swapfile seems needed on 0.2.1 it is still strange that the upgraded one (0.1.9->0.2.1) works without

roleoroleo, 2022-03-22:
My camera never worked without swap files.
But it really depends on the configuration.

closed 2023-01-08 by stale[bot]