roleoroleo/yi-hack-Allwinner-v2 #399: Yi Dome Guard YRS r35gb does not connect to wifi after changing credentials multiple times

vielfarbig, 2022-03-30:
My Yi Dome Guard YRS r35gb does not connect to wifi after I changed the credentials a 2nd time with the wiki entry https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials.

Initially I hacked the camera without the app and changed the wifi credentials afterwards. I worked, i had access to its website. Great work from you!

But I tried to change the wifi credentials for a different wifi, but it would not connect. The credentials should be correct, have no unallowed chars and it works with other devices. The camera also would not connect to the first wifi any longer.

So I thought, i could flash it again. I deleted all files from the sd card and tried the README-guide again. It seems that it also ran (there are logs in the logs folder with a newer timestamp), so it seems that the camera is not completly bricked (?)

This does not help, the camera can still not connect to the wifi. Is it bricked? It seems to not boot loop. So I want to ask first, if i should try to unbrick it or what I might get wrong and how to fix.

I don't know what you need for debugging, but I will try to provide it!

Thank you in advance for your help!

roleoroleo, 2022-03-31:
Try to use the back button to reset the cam.

vielfarbig, 2022-04-01:
Yes, that worked. I can connect to the first wifi again.

But I still cannot change it to the other wifi. It must be something with the wifi (a different one works).
The problematic wifi is configured by an old router being in bridged mode with normal wifi, dhcp, wpa2 etc. I can connect to it with other devices.

How can i debug that?

roleoroleo, 2022-04-01:
Probably playing with wpa_supplicant.
But I'm not an expert.

vielfarbig, 2022-04-07:
**tldr**: The udhcpc seems to not get a lease somehow. I will use the LAN with a persistent fix as a workaroud.

The LAN also does not work out of the box. So I tried to get it running as as workaround for wifi (in my case i would also be fine, because the camera should be a fixed place and it should be more reliable)  

I changed the wifi to a one that works (Fritz Box 7490) and tried to get LAN by executing `udhcpc -i eth0 -b -s /home/app/script/default.script` as described in https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/34#issuecomment-762325612 That gets a lease on eth0:
```
root@yi-af3e:~# udhcpc -i eth0 -b -s /home/app/script/default.script
udhcpc: started, v1.27.2

Setting IP address 0.0.0.0 on eth0
udhcpc: sending discover
udhcpc: sending select for 192.168.178.102
udhcpc: lease of 192.168.178.102 obtained, lease time 864000
Setting IP address 192.168.178.102 on eth0
 need update route and dns with eth0
Deleting routers
Adding router 192.168.178.1
Recreating /tmp/resolv.conf
 Adding DNS server 192.168.178.210
Set DNS config
```

But this is not persistent, so i appended it to the `startup.sh` script as described in https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials If the yi is connect via lan before booting, gets a lease after starting. 

So i will use the lan option as a workaround, but i also tried to figure out whats with the wifi.
If i execute first
```
killall wpa_supplicant
killall wpa_supplicant
ifconfig wlan0 down
```
and than wifidhcp, udhcpc does not get a lease:
```
root@yi-af3e:/home/app/script# ./wifidhcp.sh 
killall: udhcpc: no process killed
udhcpc: started, v1.27.2

Setting IP address 0.0.0.0 on wlan0
udhcpc: sending discover
udhcpc: sending discover
udhcpc: sending discover
udhcpc: no lease, forking to background
```

 So i did a scan on wlan0, but i cannot a magnificient difference between the working wifi (Cell 01) and the non working (Cell03):
```
iwlist wlan0 scan

wlan0     Scan completed :
          Cell 01 - Address: XX:XX:XX:XX:XX:XX
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=70/70  Signal level=-27 dBm  
                    Encryption key:on
                    ESSID:"Fritz Box 7490"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s; 6 Mb/s
                              9 Mb/s; 12 Mb/s; 18 Mb/s
                    Bit Rates:24 Mb/s; 36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000XXXXXXXXX
                    Extra: Last beacon: 1536ms ago
                    IE: Unknown: [...]
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: XXXXXXXXXXXXXXXXXXXX

          Cell 03 - Address: XX:XX:XX:XX:XX:XX
                    Channel:1
                    Frequency:2.412 GHz (Channel 1)
                    Quality=69/70  Signal level=-41 dBm  
                    Encryption key:on
                    ESSID:"WLAN-XXXXXX"
                    Bit Rates:1 Mb/s; 2 Mb/s; 5.5 Mb/s; 11 Mb/s
                    Bit Rates:6 Mb/s; 9 Mb/s; 12 Mb/s; 18 Mb/s; 24 Mb/s
                              36 Mb/s; 48 Mb/s; 54 Mb/s
                    Mode:Master
                    Extra:tsf=0000000XXXXXXXXX
                    Extra: Last beacon: 1444ms ago
                    IE: Unknown: [...]
                    IE: IEEE 802.11i/WPA2 Version 1
                        Group Cipher : CCMP
                        Pairwise Ciphers (1) : CCMP
                        Authentication Suites (1) : PSK
                    IE: Unknown: XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX
                    IE: Unknown: XXXXXXXXXXXXXX
```

The working `wpa_supplicant.conf` looks like the following:

```
root@yi-af3e:/home/app/script# cat /tmp/wpa_supplicant.conf 
ctrl_interface=/var/run/wpa_supplicant
ap_scan=1
network={
ssid="Fritz Box 7490"
scan_ssid=1
proto=WPA RSN
key_mgmt=WPA-PSK
pairwise=CCMP TKIP
group=CCMP TKIP
psk="XXXXXXXX"
}
```

roleoroleo, 2022-04-13:
At the moment, you are using the same channel on both routers.
Maybe the Fritz Box is too powerful.

vielfarbig, 2022-04-17:
Yes this could be a problem. But it works with other devices and they don't show the Fritz Box when connecting with the other router.
Possibly the old router sends some metadata that udhcpc rejects?

vielfarbig, 2022-04-17:
So I don't need the connection via the old router. But I can perform actions if that helps you to identify some more general issues.

roleoroleo, 2022-04-17:
> Possibly the old router sends some metadata that udhcpc rejects?

I don't know but it should ignore unknow parameters.

closed 2023-01-08 by stale[bot]