roleoroleo/yi-hack-Allwinner-v2 #401: Connection refused from HA since last firmware update

PilaScat, 2022-04-01:
I'm facing this problem after the latest update, I don't know if it's the cam or the integration
[image]

Sarunas88, 2022-04-05:
same, one camera out of 4 does this. done resetting and integrating everything from scratch, nothing helps. connection refused and thats all.

Sarunas88, 2022-04-05:
[image]

Sarunas88, 2022-04-05:
[image]

roleoroleo, 2022-04-07:
I will release 0.2.2 asap.
Check if it works.

PilaScat, 2022-04-07:
ok thanks

Sarunas88, 2022-04-16:
Failed to get system configuration from device 192.168.0.68: error HTTPConnectionPool(host='192.168.0.68', port=8080): Max retries exceeded with url: /cgi-bin/get_configs.sh?conf=system (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7f8b0dadd8b0>: Failed to establish a new connection: [Errno 113] Host is unreachable'))

Unable to get configuration from the cam

Error from stream worker: Error opening stream (ERRORTYPE_111, Connection refused) rtsp://****:****@192.168.0.105/ch0_0.h264

Error from stream worker: Stream ended; no additional packets
Error from stream worker: No dts in 7 consecutive packets

Error getting status of device 192.168.0.105: error HTTPConnectionPool(host='192.168.0.105', port=8080): Max retries exceeded with url: /cgi-bin/privacy.sh?value=status (Caused by ConnectTimeoutError(<urllib3.connection.HTTPConnection object at 0x7f8b0e683df0>, 'Connection to 192.168.0.105 timed out. (connect timeout=10)'))

Failed to get status on device 192.168.0.105: error unknown

Error from stream worker: Stream ended; no additional packets

Sarunas88, 2022-04-16:
only 1 out of 3 Cameras are running. 3 failed with same results :(

dreimer1986, 2022-04-16:
Since yesterday the cams stop working out of nowhere. 2 of 2. 
[image]
This is what I see when this happens.
Some reboots later they work for a while again

dreimer1986, 2022-04-18:
Maybe it's due to me activating swap or the new version. But they are running for 9 hours now which is a record when you look at the last week.

Sarunas88, 2022-04-22:
so i got everything up to date and tried lots things, and one helped me, the cameras are running again. so i think i had very difficult password set on my cameras !? i changed it to something more easily 4 digits 4 letters and the cams are running again :/ we'll see for how long.

dreimer1986, 2022-04-25:
Mine broke down again. This time only one. On Saturday both were gone. After a forced reboot they come back though:

[image]

roleoroleo, 2022-04-25:
I have a doubt.
Did you configure a reservation for your cam? It is necessary.
Maybe, the cam changes ip address and it becomes unreachable.

Sarunas88, 2022-04-25:
they are always the same, it got better woth the shorter password but didn't really changed much. the streams are vwry lagging and brakeup often

dreimer1986, 2022-04-25:
> I have a doubt. Did you configure a reservation for your cam? It is necessary. Maybe, the cam changes ip address and it becomes unreachable.

Yes I did. They completely vanish off the network. After a hard reset they are back on duty.

edmundasa, 2022-04-29:
I have same issue with 0.2.2 software.
Live stream and snapshot doesn't even load.
top load shows 8-9 ranges constantly. RMM CPU usage increased around 10% from 0.2.1 version with default settings.
On 0.2.1 versions top shows load around 4

roleoroleo, 2022-05-05:
I tried to investigate the problem but I did not find anything.
I need to know which elements of the integration are not available.

- rtsp stream?
- still images?
- sensors?

> After a hard reset they are back on duty.

Reset of the cam? Whit the back button?
Or reset to factory from the web page?

> top load shows 8-9 ranges constantly. RMM CPU usage increased around 10% from 0.2.1 version with default settings.

RMM is a yi process. It is very unlikely that I can do anything that changes the cpu load of this process.

edmundasa, 2022-05-06:
Hi @roleoroleo to upgrade I used automatic update and after that RTSP snapshots weren't loading and web interface was very laggy.
After that I reset camera, re-formated sd card and manually installed 0.2.2 and all of issues are gone. So 0.2.2 in itself is working as it should. Thanks. 
Maybe upgrade from interface messed up something in my camera (r35gb).

Sarunas88, 2022-05-06:
Me too, i have no problems anymore. It took some time after the HA Update and 0.2.2 but I have no problems anymore. do not know why that suddenly working again.

roleoroleo, 2022-05-07:
The ota procedure is complicated and it can happen that something goes wrong.

TheBisk, 2022-05-31:
same problem here, 2 of 4 cam not working:
Error from stream worker: Error opening stream (ERRORTYPE_111, Connection refused) rtsp://****:****@192.168.xxx.xxx:554/primary
Error from stream worker: Stream ended; no additional packets

RadaKk, 2022-06-15:
Same for me too, I updraded to 0.2.2 manually and I could not access any of the RTSP channels from my dome guard.
Try to revert to 0.2.0 but I face the same issue now...

RadaKk, 2022-07-21:
Is there any update on this issue? I tried to switch back to each available version of the hack for my dome guard and none of them allow me to use RTSP, snapshot... But everything is working through the yi-home app.

dreimer1986, 2022-07-22:
Well after I posted here my cams started to behave as I expect them to do. All working fine.

fredmassa, 2022-08-05:
> 

do you used the "Reset to yi-hack default" button via webpage?

fredmassa, 2022-08-05:
> 

Hi @edmundasa , do you used the "Reset to yi-hack default" button via webpage?

edmundasa, 2022-08-05:
@fredmassa no, just formated sd card and put newest hack version files there.

fredmassa, 2022-08-07:
> @fredmassa no, just formated sd card and put newest hack version files there.

Thanks @edmundasa but no good news... formatted but I've always the same error message:
_2022-08-07 11:05:47.714 ERROR (stream_worker) [homeassistant.components.stream.stream.camera.cam3] Error from stream worker: Error opening stream (ERRORTYPE_111, Connection refused) rtsp://xx.xx.xx.xx:554/ch0_0.h264_

any idea @roleoroleo

fredmassa, 2022-08-07:
I tried also opening it form the internal page using vlc but nothing
[image]

edmundasa, 2022-08-07:
could it be your network issue? some router blocking?

fredmassa, 2022-08-07:
I don't think.. i have 3 cam, 1 with yi-hack-allwinner and 2 with yi-hack-Allwinner-v2. They were working since 2 months ago and now, same config, same settings, same network.., only the one with yi-hack-allwinner is working properly. The other 2 reply me with that error

fredmassa, 2022-08-07:
I can turn on/off them but not stream

roleoroleo, 2022-08-16:
Did you run some yi fw update?
What damon are you running? Standard or alternative?

To debug the problem, please remove ha from the chain: try to read the stream with vlc (try 3-4 times).
If it doesn't work open a ssh terminal and check if the process is running:
`ps | grep rRTSPServer`

fredmassa, 2022-08-17:
If I try to use the default password, I mean leaving it empty, I cannot login via ssh

roleoroleo, 2022-08-18:
Please post your config (hide personal data).
`cat /tmp/sd/yi-hack/etc/system.conf`

fredmassa, 2022-08-18:
yes, here the configs:
**CAM2**
[image]

**CAM3**
[image]

roleoroleo, 2022-08-18:
RTSP_AUDIO field is empty, it's not normal.
Try to set a value.

fredmassa, 2022-08-18:
I set the default (disabled) but nothing changed
[image]

roleoroleo, 2022-08-20:
The file is ok, I don't understand.

roleoroleo, 2022-08-20:
You should try to restart from scratch.
Reset to default the cam with the button.
Then format the sd card and apply again the hack.

fredmassa, 2022-08-23:
Ok! Please give me 1 week and half and I'll update you

fredmassa, 2022-09-10:
Ok.. I've done the sd format and the camera hard reset. Followed the hack instruction (pairing with yi app and hack) and the cam was sending the video properly to HA. After 10 minute I moved the cam to the right room and unfortunately I had again the error message. Moved again in the previous position (the one that worked) I have the same error message.

ghanashyaml, 2022-09-10:
Maybe something to do with internet availability in that area and something
goes wrong after that?

fredmassa, 2022-09-10:
I don't think so because using yi app it's all working properly

fredmassa, 2022-09-19:
@roleoroleo 🥳🥳🥳
After last updates all the cams are working properly!!!!

closed 2022-12-18 by roleoroleo