roleoroleo/yi-hack-Allwinner-v2 #404: How to apply the change after /cgi-bin/camera_settings.sh?conf=value?

outofsight, 2022-04-03:
After changing a camera setting with the api `/cgi-bin/camera_settings.sh?conf=value`, for example after a `GET` to turn off the led, `http://{{address}}:8080/cgi-bin/camera_settings.sh?led=no`, the setting is not immediately applied. 
 
I can see the changed setting in the web page, ma the led on the camera is still on.

How to apply che change?

outofsight, 2022-04-04:
Found the issue. `LED` and other settings key should be lowercase. I was sending uppercase because from `http://ip:8080/cgi-bin/get_configs.sh?conf=camera` you get the settings uppercase, and I was confused because the page `http://ip:8080/index.html?page=camera_settings` was correctly updating the settings.

roleoroleo, 2022-04-04:
Maybe I could ignore the case...

closed 2023-01-08 by stale[bot]