roleoroleo/yi-hack-Allwinner-v2 #406: Doom Guard r30gb KFUSY onvif stop to work

rt400, 2022-04-05:
Hi,
Firmware Version	0.2.1
Base Version	9.0.05.33_202009251856
Model Suffix	r30gb
Serial Number	KFUSY36N7H05FK200808

the rtsp work somtime but onvif never work.
i try with Onvif Device Manager but i get no signal. also try it with HIKVISION  NVR and also no signal.
the camera connect to the LAN and the onvif interface set on Network.
any idea ??

roleoroleo, 2022-04-07:
Check if the onvif_srvd process is running.
And check the configuration file in /tmp.

rt400, 2022-04-09:
i dont know which command to wrote to check what the services are running.
the onvif_srvd.conf are not empty and its looks like that is config correct

roleoroleo, 2022-04-16:
Use "ps" command to check if onvif_srvd is running.
Use "cat /tmp/onvif_srvd.conf" command and post the output.

rt400, 2022-04-16:
> 

```pid_file=/var/run/onvif_srvd.pid
model=Yi Hack
manufacturer=Yi
firmware_ver=0.2.1
hardware_id=KFUS
serial_num=KFUSY36N7H05FK200808
ifs=eth0
port=80
scope=onvif://www.onvif.org/Profile/S

#Profile 0
name=Profile_0
width=1920
height=1080
url=rtsp://%s/ch0_0.h264
snapurl=http://%s:8080/cgi-bin/snapshot.sh?res=high
type=H264

#Profile 1
name=Profile_1
width=640
height=360
url=rtsp://%s/ch0_1.h264
snapurl=http://%s:8080/cgi-bin/snapshot.sh?res=low
type=H264

user=admin
password=XXXXXX

#PTZ
ptz=1
move_left=/tmp/sd/yi-hack/bin/ipc_cmd -M left
move_right=/tmp/sd/yi-hack/bin/ipc_cmd -M right
move_up=/tmp/sd/yi-hack/bin/ipc_cmd -M up
move_down=/tmp/sd/yi-hack/bin/ipc_cmd -M down
move_stop=/tmp/sd/yi-hack/bin/ipc_cmd -M stop
move_preset=/tmp/sd/yi-hack/bin/ipc_cmd -p %t
```

rt400, 2022-04-16:
the onvif service is running..

roleoroleo, 2022-04-17:
The config file is ok.
When you write that the cam is connected in LAN you mean ethernet. Right?

> the onvif service is running..

Did the problem disappear?

rt400, 2022-04-17:
yes its connect in ethernet (LAN). the problem still exist...

roleoroleo, 2022-04-17:
Try to kill the service and run it manually:
```
. /tmp/sd/yi-hack/script/env.sh
onvif_srvd --no_fork --no_close --conf_file /tmp/onvif_srvd.conf
```

rt400, 2022-04-17:
still not work

rt400, 2022-04-17:
i set the onvif in wireless and still not work

roleoroleo, 2022-04-18:
Any message in console when you run the process manually?

rt400, 2022-04-18:
No..

rt400, 2022-05-25:
@roleoroleo any idea how to fix it ?

roleoroleo, 2022-05-25:
If I understand correctly Onvif Device Manager connects successfully to the cam and is able to retrieve all settings, profiles, etc...
But the live stream doesnt' work (no signal).
If this is behavior, onvif is running properly. The problem is the rtsp daemon.
Did you set any password for rtsp?
If you try to read the stream with vlc, does it work?

rt400, 2022-05-25:
yes i set password for rtsp also from the vlc its work fine

roleoroleo, 2022-05-25:
Try again ODM and set your credentials before to connect.
When you select live stream button, it may take a few seconds.

rt400, 2022-05-30:
Still my NVR cannot get stream from onvif , only that Camera. i have 3 (y20ga, r30gb, r35gb) all setting are the same and the onvif work fine only r30gb not get the stream

roleoroleo, 2022-05-30:
If I remember correctly, r30gb streams a h265 stream.
Maybe your nvr is only h264 compatible.

rt400, 2022-05-30:
can i change it to h264 ?

roleoroleo, 2022-05-30:
There is an experimantal procedure:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/issues/98

rt400, 2022-05-30:
i set the onvif to low and now its work, guess you are right about the H265.
cant understand how really change the h265 to h264, hope you can make for me a version in high quality h264 .

roleoroleo, 2022-05-31:
To switch h265 to h264 you need to change a byte in the partition where configuration is stored.
It's a risky task.

rt400, 2022-05-31:
OK. I close this issue cause I mange it in different way...

closed 2022-05-31 by rt400