roleoroleo/yi-hack-Allwinner-v2 #407: It is possible to change camera settings via MQTT?

outofsight, 2022-04-05:
It is possible to change camera settings via MQTT?

I enabled the MQTT config topic, say `yicamera/yi-cam-04/config`.
I publish (from MQTT Explorer) `yes` under the topic like `yicamera/yi-cam-04/config/camera/LED` but nothing changes.
I tried several combination of casing.

roleoroleo, 2022-04-07:
It doesn't work.
You should try to use mqtt advertise:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/src/static/static/yi-hack/script/mqtt_advertise

itkfilelor, 2022-04-18:
How do we set this up?
Pardon my noobness =)

roleoroleo, 2022-04-18:
You should enable the config section:
[image]

and send the following command:
Topic:
YOUR_MQTT_TOPIC_PREFIX/camera_setting/LED/set
Payload:
yes or no

itkfilelor, 2022-04-20:
Oof, didn't see the link at the top of the MQTT page. :sweat_smile:

closed 2022-10-26 by roleoroleo