roleoroleo/yi-hack-Allwinner-v2 #408: MQTT motion_files message has empty files []

marcospg75, 2022-04-06:
Hello,
When a motion detection happen, I receive a motion_files topic with the start and end times and the related files recorded but this files are always empty. But if you check with the web service, you can find them recorded in the SDCard at the camera

Firmware: 0.2.1
Camera: 1080p HOME AI+ BFUS

roleoroleo, 2022-04-07:
I know, the problem is the timezone.

marcospg75, 2022-04-07:
> I know, the problem is the timezone.
> 
> EDIT
> 
> I fixed the problem, next release...

Great! thank you very much!!

marcospg75, 2022-04-08:
Any workaround while you develop next release?

roleoroleo, 2022-04-13:
No, is binary code.
But I'm almost done.

roleoroleo, 2022-05-20:
Fixed here:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/9cfe97ad89196519237091779dde3663f6d9e066

jjcasmar, 2022-05-20:
Is this patch already in a release? Or should I compile master myself?

jjcasmar, 2022-05-20:
I see it is in 0.2.2, which is the version I have in my camera, but I still receive an empty list of files

roleoroleo, 2022-05-20:
Check if your configuration is ok.
Set the timezone both in the app and in the webpage.

jjcasmar, 2022-05-20:
afaik, its is correct., in the app and in the webpage, and the timestamp inside the mqtt payload is correct

roleoroleo, 2022-05-20:
Please, send me another example of the json message.

jjcasmar, 2022-05-20:
```{
  "start": "2022-05-20T18:06:34+02:00",
  "end": "2022-05-20T18:07:35+02:00",
  "files": []
}```

The timestamps are correct, I in Madrid (CET-1CEST,M3.5.0,M10.5.0/3)

roleoroleo, 2022-05-21:
Are there files?
/tmp/sd/record/2022Y05M20D16H/06MxxSxx.mp4
/tmp/sd/record/2022Y05M20D16H/07MyySyy.mp4

jjcasmar, 2022-05-21:
```root@door_camera:/tmp/sd/record/2022Y05M20D16H# pwd
/tmp/sd/record/2022Y05M20D16H
root@door_camera:/tmp/sd/record/2022Y05M20D16H# ls
E206M36S24.jpg  E206M36S24.mp4  E207M02S58.jpg  E207M02S58.mp4
```

roleoroleo, 2022-05-21:
Your cam uses different format for file names.
I need to add some line of code.

closed 2023-01-08 by stale[bot]