roleoroleo/yi-hack-Allwinner-v2 #412: Hack no longer persistent through reboots - Dome Guard IFUS

dieideeistgut, 2022-04-14:
Hello!

First up i'd like to say a million thanks for the work @roleoroleo and many others have put into this. I've been quite happy running Yi cams since a long while now. But the last some days is stumbled upon an issue on the only Dome Guard Cam i have. Been rocking the r30gb ever since on this cam. Up until last week this cam was hooked directly over ethernet to my home server but coming home one day i realised the cam was no longer in the network. Rebooting greeted me with the "sweet" Cam voice telling me how welcome i am to Yi home cameras...

So i started several things:

- Un-Hack & Re-Hack
- Un-Hack -> Un-Brick with the new procedure -> Re-Hack

and so on. The Hack itself is working now again, MQTT and all other functions are available and working nicely but: I have to do this stupid setup routine all the time. Whenever i reboot i have to do the init-process aka show the cam a QR code, etc, pp. After that everything is fine up until the next reboot. 

I'd appreciate any further suggestion and thanks a ton again for this awesome world of yi-hacks.

roleoroleo, 2022-04-17:
Probably the cam fails to save the wifi settings.
You could try to follow this guide:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials

dieideeistgut, 2022-04-26:
Thanks for your reply and hint. Came a little further with this and i may have found the issue. Setting WiFi Info via dd seems not to work at all anymore. I tried the same process on two cameras:

dd does not write valid data anymore. setting the ssid and key results in a different response on my not-persistant-camera. The ssid and key strings get altered so i assume they are not valid. As a last measurement i dd'd my block7 from my original backup back to the cam. But this too seems not to write valid strings anymore. I assume i bricked the cam hardware at some point. 

So this can be marked as resolved ;)

closed 2022-04-26 by dieideeistgut