roleoroleo/yi-hack-Allwinner-v2 #413: Swapfile not working

Netlims, 2022-04-14:
Since version 0.20 the swapfile option seems to be ignored.
After I upgraded my camera keeps rebooting every few minutesand when I ssh into it and run 'top' I can see no difference whether I'm using swapfile or not.
I tried creating and using the swapfile myself but many commands are not known to busybox.
How can I do that?

My camera model is h60ga and I'm currently using version 0.21

Thanks a lot!

roleoroleo, 2022-04-17:
To use all the command you have to sources the environment:
```
. /tmp/sd/yi-hack/script/env.sh
```

Run "free" command and post the output.

Netlims, 2022-04-17:
[image]

I'm on vacation right now so I don't have access to the web interface, only ssh.
If there is anything else you need please let me know.
Do you recommend I will enable swap by myself or should we fix the internal function?

roleoroleo, 2022-04-17:
Your swap file doesn't work.
You should check your sd card.
- Remove it from the cam (when you are at home).
- Insert it on a PC and run a file system check.
- Remove the file "swapfile" from the card.
- Check if you have enough space (64 MB).

If this doesn't work, probably there is a problem with your model.

Netlims, 2022-04-24:
My PC indeed detected issues with the SD card, I fixed the issues and deleted the swapfile as suggested.
The camera recreated the swapfile and now everything work!
Thank you very much!
I really appreciate your work and your help

closed 2022-04-24 by Netlims