roleoroleo/yi-hack-Allwinner-v2 #414: Home 1080p boots only when connected to UART

onfoot, 2022-04-15:
For a while the cam was working fine with the hack, but at one point just stopped booting, with only the orange light on when powered. I finally decided to unbrick it the hard way, since I had no backup (of course).

So I opened it up, soldered the UART pads, and powered it on. I started watching the logs to see any indications of failures, ready to screw with the uboot to recover the firmware from my second cam, and to my surprise, it booted up normally. I did a few power cycles to make sure, updated the hack to the latest version, all was good. However once started it while not connected to UART, it wouldn't boot again. Made sure it's actually the case and indeed, every time it's powered while connected to UART, everything is fine.

Anything I could do to fix this?

roleoroleo, 2022-04-17:
Are you using a different power adapter or cable?
Try the original.

onfoot, 2022-04-21:
I tried a few different adapters and cables, nothing. Doesn't boot up until connected to a UART.

onfoot, 2022-10-07:
Well, what do you know. I dropped this issue for a while, thinking the cam is a lost cause, and I bought two more because I apparently wasn't miserable enough. I connected the "broken" one using a new cable and it magically works. 🎉

closed 2022-10-07 by onfoot

roleoroleo, 2022-10-07:
Unbelievable...

onfoot, 2022-10-09:
For posterity's sake, when using a different cable, taping the usb data pins with kepton or electrical tape, on the big plug side, helps. I needed a short cable for one particular spot. 👍