roleoroleo/yi-hack-Allwinner-v2 #415: Outdoor 1080 YHS.3020

pashok2398, 2022-04-17:
Hi All,
I am tying to flush my cam Outdoor 1080 YHS.3020 Firmware 9.0.26.07_202104231123
I tried 2 different SD card and format it with camera and with PC.
I also tried the latest and the 0.1.8 release `h30ga`.
The directory `factory` does not renamed and the yellow light keep on (no blinking and no blue).
What am i missing?
Thanks

roleoroleo, 2022-04-17:
What operating system are you using to create the sd card.
Please note that factory is upper case: Factory.
If you try to make a backup, does the procedure work?

pashok2398, 2022-04-18:
@roleoroleo 
Good catch, I use Win10
Thanks a lot, I re-format it using my F35 and its worked like a charm...
Thanks!!!

closed 2022-04-18 by pashok2398