roleoroleo/yi-hack-Allwinner-v2 #419: Kami 1080p Home BFUS with 9.0.20

Bambuspony, 2022-04-21:
Great work in general! Using the allwinner for some time now. The new cameras are from Kami with higher firmware (9.0.20.06_202007061841).
y20ga (allwinner) is not working
y21ga (allwinner2) keeps in bootloop even with swap
y211ga - confguration is available but no rtsp or onvif
y28ga (allwinner2 Kami mini) rtsp works but onvif is not available
Is it possible to get the onvif working in some way?

roleoroleo, 2022-04-23:
This model is not supported at the moment.
But if your firmware is 9.0.20 you should use 28ga.

Did you test onvif with Onvif Device Manager?

Bambuspony, 2022-04-25:
oh, used Ip Centcom for observation. With Onvif Device Manager the cam is available and onvif ist working.
The cam is also live again, so I can confirm that y28ga is working with Kami 1080p BFUS.

closed 2023-01-08 by stale[bot]