roleoroleo/yi-hack-Allwinner-v2 #421: Yi Outdoor IFUS - Firmware Version - 0.2.2

logzorz, 2022-04-22:
Hi guys, I've recently flashed my outdoor camera with the latest firmware, I have an issue where I cannot access the status page and it appears to spike the ping to 1300ms, aswell as the image quality coming through with lines across the live image, the camera ran fine prior for about 6 hours. Im unsure on how to send you guys a log if you could please advise.

Thank you.
[image]
[image]

roleoroleo, 2022-04-23:
You should try to restore the original fw and check it the cam works.
I think it's a hardware problem.

logzorz, 2022-04-23:
I think you may be right, I reflashed the Y21GA firmware as a test and its doing the same thing. Ill restore the original firmware and report back. Thank you for the reply Roleoroleo!

logzorz, 2022-04-23:
Confirmed hardware problem! Apologies for the issue raise!

closed 2022-04-23 by logzorz