roleoroleo/yi-hack-Allwinner-v2 #422: yi home 1080p no record on sd card

kidlerz, 2022-04-22:
Hello,
Base Version | 9.0.19.02_202007141512
Model Suffix | y21ga

Disable Cloud  - ON
Recording without Cloud - ON
Switch on/off the camera - ON
Save video when a motion is detected - OFF

and no files on sd card /tmp/sd/record/

why ?

razem-io, 2022-06-21:
Same here. It only works for me if I am connected to Wi-Fi. Will try every option and report back if anything works.

roleoroleo, 2022-06-28:
> Same here. It only works for me if I am connected to Wi-Fi. Will try every option and report back if anything works.

Is the cam connected to the network?

razem-io, 2022-07-12:
Hey @roleoroleo, somehow missed the notification, sorry. 

Yes, the cam is connected to the network. Then the recording works without issues. If I take it somewhere else, where the wifi is not available, it won't record any videos.

Also my model is the QFUS, so I am using y211ga not y21ga.

razem-io, 2022-07-15:
@roleoroleo if it is time issue I can try to fix it myself. I was browsing through the project to find where it could fail. So far I haven't found anything. Do you have a suggestion?

roleoroleo, 2022-07-15:
Other people have reported this problem.
Unfortunately it depends on the yi processes and I have not found a solution.

razem-io, 2022-12-18:
Not stale, just no solution yet.

closed 2022-12-18 by kidlerz

razem-io, 2022-12-18:
@kidlerz why did you close the issue? It still exists, right?