roleoroleo/yi-hack-Allwinner-v2 #427: system.conf readonly

cbroissa-github, 2022-05-02:
Hello,
I made the update, my cam works but I can't modify the configuration from myIP:8080, when I update, the options reset as before.
With SSH, the file is readonly /tmp/sd/yi-hack/etc/system.conf [Readonly] 
Reboot change nothing.

There is a solution apart from erasing the SD card ?

WalkingDot, 2022-05-03:
I'm on a older version, so I can't test it. Maybe this helps:
`chmod u+w system.conf`

Than it should be:
`-rwxr-xr-x`

cbroissa-github, 2022-05-03:
No I can't change rights

root@(none):/tmp/sd/yi-hack/etc# chmod u+w system.conf
chmod: system.conf: Read-only file system

Staini, 2022-05-03:
I think roleo can help you if you post your camera model. It's a mounting problem probably.

cbroissa-github, 2022-05-04:
Yi Outdoor 1080p IFUS
Base version : 9.0.26.07_202104231123
Firmware : 0.2.2

roleoroleo, 2022-05-05:
I think that the sd card is corrupted and the filesystem is mounted read-only.
Extract the card from the cam and fix it with a file system check on your pc.
Probably there will be corrupted files.