roleoroleo/yi-hack-Allwinner-v2 #428: ESCAM PT202	RTSP NOT WORKING

tv4you2016, 2022-05-02:
Hi
I'm not getting the rtsp to work on my ESCAM PT202. 

* Firmware version: 0.2.2 
-  FTP: OK
- [ OK] SSH 
- [ OK] Snapshot
- [ OK] ONVIF
- [ WIFI] Wifi/ETH

[image]

[image]

[image]

orion131313, 2022-05-02:
I had the same problem.
I activated the "RTSP alternative server" and it started working.
However I noticed that after a few hours vlcplayer stops displaying the images. If I restart the camera it works again until it blocks again.
I think it's a firmware error.

roleoroleo, 2022-05-05:
The standard RTSP server should work.
It comes from a very popular and very stable library (live555).
Alternative server is simpler and rougher, but it's more compatible with some hardware nvr.

So, you should use the first one.

Please, when you try to play with vlc, try 2 or 3 times. Sometimes vlc is hard to get started.
If it doesn't work, try another rtsp client, ffmpeg is good test.
`ffmpeg -r 20 -rtsp_transport tcp -i rtsp://192.168.1.213/ch0_0.h264 /tmp/test.h264`