roleoroleo/yi-hack-Allwinner-v2 #430: Missing timestamp

tunerooster, 2022-05-03:
I have 5 YI Outdoor 1080p cameras all running Allwinner-v2 ha30ga.  They were all purchased together at the same time, and all have serial number starting with IFUSY59.

The problem is, the latest camera, which I just deployed, does not implement a timestamp in the video.  The other 4 cameras all do implement a timestamp.

I have been trying to determine the difference between the cameras that do and the one that does not implement the timestamp.  I have found the following differences:
```
                                Cams With Timestamp        Cam Without Timestamp
Base Version:                  9.0.26.07_202104231123     9.0.26.05_202011201756
ls -l /home/app/rmm            2249956 Apr 23, 2021 rmm   1979100 Nov 20, 2020 rmm
```
The "Maintenance" tab on the web page for all cameras says:
```
Installed: 0.2.2 - Available: 0.2.2    [UPGRADE FIRMWARE]
```
The "YI App" shows the same "Firmware Version" as is shown on the Maintenance page "Base Version", as expected.  Also from the YI App, the Firmware Version indicates "Up to Date" on all cameras, even though there is clearly a newer release (9.0.26.07_202104231123) for the problem camera.  (I assume this is the cause of the missing timestamp, i.e., the older YI firmware.)

Bottom line, I can't find a way to upgrade the YI Firmware for the problem camera to the 9.0.26.07 version.

Because the /home filesystem is read-only,  I also cannot find a way to copy the newer /home/app/rmm from a working camera to the problem camera, which I have a hunch would also fix the problem.

I do not understand how the problem camera came with different firmware from the others in the first place.

Can you suggest a solution for me?

Thanks as always!

tunerooster, 2022-05-04:
A follow-on question:

Does a yi-hack new release also update yi native firmware?  I.e., would, for example, a 0.2.3 yi-hack release include the latest yi firmware?

If not, does anyone know if there is a way to manually download and install the latest yi firmware, and from where?

Thanks again!

roleoroleo, 2022-05-05:
The hack doesn't change the yi fimware and the timestamp, as you already noticed, is defined in the rmm process.

You could try to "copy" the yi fw from the 1st cam to the 2nd cam (if the cams are hardware identical).
You should dump the firmware with the procedure explained here:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Dump-your-backup-firmware-(SD-card)
And you should restore the firmware in the 2nd cam with the same procedure used while unbricking:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/tree/master/unbrick

It's not simple to know which partitions you need to overwrite but probably home and backup are enough.

The only unexplained step is to change the names:
mtd3 -> home_h30ga
mtd4 -> backup_h30ga

tunerooster, 2022-05-07:
Thank you for your reply.

It appears that this requires removing the sd card from the camera.  As the camera is remote and "glued in" with silicon, that is difficult to do.  If you can think of any way to fix it without removing the sd card it would greatly help.

I should mention...

With the problem camera, I installed the yi-hack software on the sd card *before* even pairing the camera for the first time.  There were no apparent problems as it came up without a hitch.  But I do not believe I did this with the other cameras (i.e., I paired them first before ever inserting an sd card).  Is it possible this caused the issue?

If I *have* to remove the camera (i.e., to get the sd card out), couldn't I just reset and re-pair it without inserting the sd card?  I am thinking that doing this would cause the yi firmware to recognize that it's firmware is out of date, and therefore allow a yi firmware OTA update.  It is only the yi firmware that is out of date.

You can't think of any way to trick the yi firmware into realizing it is outdated, can you?

Thanks so much for your advice and help!

tunerooster, 2022-05-07:
I am trying to understand the "unbrick" procedure you refer to above...

Step 6 says: Unzip it in the root folder of your cam.

Is this referring to "/" or "/root"?

In either case,  / or /root are "read-only".  So what does this mean?  I must be missing something...

Thanks!

roleoroleo, 2022-05-10:
> It appears that this requires removing the sd card from the camera. As the camera is remote and "glued in" with silicon, that is difficult to do. If you can think of any way to fix it without removing the sd card it would greatly help.

You can dump the partitions while the cam is running.
Using dd command:
```
dd if=/dev/mtdblock0 of=/tmp/sd/backup/mtdblock0.bin
dd if=/dev/mtdblock1 of=/tmp/sd/backup/mtdblock1.bin
dd if=/dev/mtdblock2 of=/tmp/sd/backup/mtdblock2.bin
dd if=/dev/mtdblock3 of=/tmp/sd/backup/mtdblock3.bin
dd if=/dev/mtdblock4 of=/tmp/sd/backup/mtdblock4.bin
dd if=/dev/mtdblock5 of=/tmp/sd/backup/mtdblock5.bin
dd if=/dev/mtdblock6 of=/tmp/sd/backup/mtdblock6.bin
dd if=/dev/mtdblock7 of=/tmp/sd/backup/mtdblock7.bin
```

About the upgrade, it is possible to make it without removing the sd card, but we have to create a new custom procedure.

> With the problem camera, I installed the yi-hack software on the sd card _before_ even pairing the camera for the first time. There were no apparent problems as it came up without a hitch. But I do [...]

I think this is not a problem.

> If I _have_ to remove the camera (i.e., to get the sd card out), couldn't I just reset and re-pair it without inserting the sd card? I am thinking that doing this would cause the yi firmware to [...]

Did you remember to enable cloud before checking the update in the app?

tunerooster, 2022-05-11:
> You can dump the partitions while the cam is running. Using dd command:
>
> ```
> dd if=/dev/mtdblock0 of=/tmp/sd/backup/mtdblock0.bin
> dd if=/dev/mtdblock1 of=/tmp/sd/backup/mtdblock1.bin
> dd [...]
Oh, that is very interesting!  Doesn't that mean you could go the other way too?  I.e., on the problem machine:
```
dd if=/tmp/sd/backup/mtdblock3.bin of=/dev/mtdblock3
dd if=/tmp/sd/backup/mtdblock4.bin of=/dev/mtdblock4
```
If that is the case, couldn't I just dd from a "good" machine" and dd the image (like above) back onto the problem machine?  I am afraid to actually try it for fear of breaking things.  Please advise if I should try this?

> About the upgrade, it is possible to make it without removing the sd card, but we have to create a new custom procedure.

I will try to help with this if you could provide an outline...

> Did you remember to enable cloud before checking the update in the app?

You mean the switch in the yi-hack web menu titled "Disable Cloud"?  I never touched that and it is currently "off", meaning cloud is enabled, right?  Or are you referring to something else?

roleoroleo, 2022-05-13:
> If that is the case, couldn't I just dd from a "good" machine" and dd the image (like above) back onto the problem machine?

It depends on the filesystem type and I'm not an expert in jffs2.
I think it's better to use a different method to write the partitions.

> I am afraid to actually try it for fear of breaking things. Please advise if I should try this?

It's not easy to brick the cam but you may need to access the sd card to unlock the cam if the procedure is not successful.

> I will try to help with this if you could provide an outline...

I will try to prepare it.

> You mean the switch in the yi-hack web menu titled "Disable Cloud"? I never touched that and it is currently "off", meaning cloud is enabled, right?

Right.

tunerooster, 2022-05-15:
Thanks!  I will check it and get back asap.  I updated my Home Assistant server and some things broke.  I need to get them resolved first.

tunerooster, 2022-05-18:
Finally got HA working...  Thanks for your patience!

Looking at: f11b51e now.

Is this a patch file (ie, the page that displays when I click on the "f11b51e" link above.)?  Sorry for my ignorance here, but can you provide a link to the full *new* build.sh` file, or post it here?  I am not clear what to do with the link above...

tunerooster, 2022-05-20:
> Check the readme:

Got it...  Thanks.  I'll let you know.

tunerooster, 2022-05-21:
Here is what I have done:
```
# Update shed ("bad" camera")  firmware using yi-hack unbrick procedure:
cd  yi-hack-Allwinner-v2-master/unbrick/h30ga
# Copy mtdblock4.bin from "good" camera.
lftp ftp://root:makeheyt@driveway -e "get /tmp/sd/backup/mtdblock4.bin; bye"
cd ..  # Back to unbrick directory
./build.sh hacked # Using the new build.sh
# After build.sh is run:
ls -l h30ga
-rw-r--r-- 1 rwk  rwk     2288 May 20 20:55 backup_h30ga.gz
-rw-r--r-- 1 rwk  rwk        6 May  6 09:47 filename
-rw-r--r-- 1 rwk  rwk      315 May  6 09:47 mtd.txt
-rw-r--r-- 1 rwk  rwk  1245184 Apr 26 19:59 mtdblock4.bin
# I next plan to upload backup_h30ga.gz to the "bad" camera (shed).  However,
# I am confused on Step 6. "Unzip it in the root folder of your cam."
# The root folder of the cam is read-only, and therefore will not unzip.
# Please advise...  What am I misunderstanding?
```
I am confused on Step 6. of the unbrick instructions "_Unzip it in the root folder of your cam_."
The root folder of the cam is read-only, and therefore will not unzip there.
**What am I misunderstanding?**

tunerooster, 2022-05-21:
Also...

I noticed the ```rmm``` executable is in the ```/home``` directory on ```/dev/mtdblock3```. 
Yet the procedure calls for ```mtdblock4.bin```.  Should I be using ```mtdblock3.bin```?
You may recall it is ```rmm``` which is causing the problem (with the timestamp).

roleoroleo, 2022-05-21:
A couple of problems:
- the size of the file backup_h30ga.gz is not correct (2288 bytes, it's little). Maybe there is an error in the build.sh.
- "Unzip it in the root folder of your cam" is "Unzip it in the root folder of your sd card"; I fixed the readme.
- You need to create both mtdblock3.bin and mtdblock4.bin

tunerooster, 2022-05-22:
I now have:
```
cd  yi-hack-Allwinner-v2-master/unbrick/h30ga
# Copy mtdblock4.bin from "good" camera.
lftp ftp://root:makeheyt@driveway -e "get /tmp/sd/backup/mtdblock3.bin; bye"
lftp ftp://root:makeheyt@driveway -e "get /tmp/sd/backup/mtdblock4.bin; bye"
cd ..  # Back to unbrick directory
./build.sh hacked # Using the new build.sh
Image Name:   xiaoyi-home
Created:      Sat May 21 19:05:42 2022
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    3407872 Bytes = 3328.00 KiB = 3.25 MiB
Load Address: 00000000
Entry Point:  00000000
modprobe: FATAL: Module mtdram not found in directory /lib/modules/5.17.4-gentoo-r1
umount: h30ga/mnt: not mounted.
### Updating h30ga/backup/init.sh
cp: cannot stat 'h30ga/backup/init.sh': No such file or directory
sed: can't read h30ga/backup/tmp_init.sh: No such file or directory
sed: can't read h30ga/backup/tmp_init.sh: No such file or directory
Image Name:   xiaoyi-backup
Created:      Sat May 21 19:05:42 2022
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    1245184 Bytes = 1216.00 KiB = 1.19 MiB
Load Address: 00000000
Entry Point:  00000000
# After build.sh is run:
ls -l h30ga
-rw-r--r-- 1 root root    2288 May 21 19:05 backup_h30ga.gz
-rw-r--r-- 1 rwk  rwk        6 May  6 09:47 filename
-rw-r--r-- 1 root root 2714131 May 21 19:05 home_h30ga.gz
-rw-r--r-- 1 rwk  rwk  3407872 Apr 26 19:59 mtdblock3.bin
-rw-r--r-- 1 rwk  rwk  1245184 Apr 26 19:59 mtdblock4.bin
-rw-r--r-- 1 rwk  rwk      315 May  6 09:47 mtd.txt
```
I did not include the output of build.sh in the prior post.  I noticed this time the error message:
```
modprobe: FATAL: Module mtdram not found in directory /lib/modules/5.17.4-gentoo-r1
```
This may have occurred previously, but I failed to notice.  Could this be the cause of the "small" size of ```backup_h30ga.gz```?

Curious that ```home_h30ga.gz``` now seems to be an expected size???

I will have to rebuild my kernel to include this module.  I will update when this is done.

Thanks!

tunerooster, 2022-05-22:
**_That was the problem!_**

build.sh now displays:
```
Image Name:   xiaoyi-home
Created:      Sat May 21 19:50:09 2022
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    3407872 Bytes = 3328.00 KiB = 3.25 MiB
Load Address: 00000000
Entry Point:  00000000
2432+0 records in
2432+0 records out
1245184 bytes (1.2 MB, 1.2 MiB) copied, 0.0802488 s, 15.5 MB/s
Successfully mounted h30ga/mtdblock4.bin on h30ga/mnt
### Updating h30ga/backup/init.sh
Image Name:   xiaoyi-backup
Created:      Sat May 21 19:50:09 2022
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    1245184 Bytes = 1216.00 KiB = 1.19 MiB
Load Address: 00000000
Entry Point:  00000000
#
ls -lt h30ga
-rw-r--r-- 1 root root 1135012 May 21 19:50 backup_h30ga.gz
-rw-r--r-- 1 root root 2714131 May 21 19:50 home_h30ga.gz
-rw-r--r-- 1 rwk  rwk        6 May  6 09:47 filename
-rw-r--r-- 1 rwk  rwk      315 May  6 09:47 mtd.txt
-rw-r--r-- 1 rwk  rwk  3407872 Apr 26 19:59 mtdblock3.bin
-rw-r--r-- 1 rwk  rwk  1245184 Apr 26 19:59 mtdblock4.bin
```
I will proceed to upload these to to the root of the *SD Card* and unzip them.  Stay tuned...

tunerooster, 2022-05-22:
P.S.  What is the chance doing this will force me to retrieve the sd card from the camera (ie retrieve the camera)?

roleoroleo, 2022-05-23:
> P.S. What is the chance doing this will force me to retrieve the sd card from the camera (ie retrieve the camera)?

At the moment the tests I have done have worked 100%.
But that's only 3-4 tests.

tunerooster, 2022-05-24:
OMG it worked!

I now have the latest yi firmware and rmm *is* producing a timestamp!

You're the best!  Thanks as always!!!

tunerooster, 2022-05-29:
Since the update, I have run into a problem with mqtt...

* In the MQTT menu view under ```Topics - Birth and will messages``` I have:
```
  Birth/Will Topic Suffix: status
  Birth Message: online
  Will Message: offline
```
* With this configuration I am seeing a never ending stream of mqtt messages as follows (about 3 per second):
```
shed/status offline
shed/status online
shed/status offline
shed/status online
shed/status offline
shed/status online
shed/status offline
shed/status online
...
```
"shed" is the name of the camera.

I have tried deleting (setting to null) the above fields in the menu and rebooting, but this results in a  streem of:
```
shed/EMPTY_TOPIC offline
shed/EMPTY_TOPIC offline
shed/EMPTY_TOPIC offline
shed/EMPTY_TOPIC offline
shed/EMPTY_TOPIC offline
...

```
None of the other cameras exhibit this problem.

I am wondering if there is another partition which I should have updated along with ```mtdblock3.bin``` and ```mtdblock3.bin```?  Or if not, if there is any other way to stop the online/offline mqtt messages.

I can't turn mqtt completely off because I rely on it for several other purposes.

roleoroleo, 2022-05-30:
Check the client id of the mqtt configuration.
You can't use the same client id for 2 different cams.
If you share the client id, the cam connects and disconnects continuously.

tunerooster, 2022-05-30:
> All 5 cams have distinct client id's as showing in the mqtt view of the
> web interface.  This one is "shed".  So apparently that's not it.  Please
> advise further...
>
> Thanks!

mqttv4.conf shows:

MQTT_CLIENT_ID=shed

tunerooster, 2022-06-01:
I think I found the problem...

I went to check it out as you suggested, but before kill mqttv4, I ran a "ps" to see thet is was running and discovered that there were *two* (2) mqttv4 processes running.  I kill one of them and the online/offline stream of messages stopped!

So I rebooted, and the TWO mqttv4 processes reappeared.  I kill one and the problem went away again.

Now I just need to find what is starting 2 mqttv4 processes...  Any ideas?

tunerooster, 2022-06-01:
Some further checking has revealed, there are other duplicate processes running:
```
{crond} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/crond -c /var/spool/cron/crontabs/
{ntpd} /tmp/sd/yi-hack/bin/busybox /tmp/sd/yi-hack/usr/sbin/ntpd -p pool.ntp.org
```
among others...

It looks like system.sh is getting run twice.

What launches system.sh?

tunerooster, 2022-06-01:
I have found the following code duplicated in ```/backup/init.sh``` on "shed":
```
# Running telnetd
/usr/sbin/telnetd &

if [ -f /tmp/sd/lower_half_init.sh ];then
    source /tmp/sd/lower_half_init.sh
elif [ -f /home/app/lower_half_init.sh ];then
    source /home/app/lower_half_init.sh
else
    source /backup/lower_half_init.sh
fi
# Running telnetd
/usr/sbin/telnetd &

if [ -f /tmp/sd/lower_half_init.sh ];then
    source /tmp/sd/lower_half_init.sh
elif [ -f /home/app/lower_half_init.sh ];then
    source /home/app/lower_half_init.sh
else
    source /backup/lower_half_init.sh
fi
```
...found at the end of the script.  It appears to be the culprit as it is not duplicated on the working cameras.

I have no idea how it got duplicated, if it has something to the /home and /backup upgrade I did, or what?

I deleted the duplicate code and now the md5sum of init.sh matches the other cameras.  I rebooted and the problem is fixed...  no duplicate processes.

If you understand how this may have occurred, I would love to hear.

Thanks again for your ongoing help!

closed 2022-06-01 by tunerooster

roleoroleo, 2022-06-01:
The hack procedure appends these lines to the file.
https://github.com/roleoroleo/yi-hack-Allwinner-v2/blob/master/sdhack/h30ga/Factory/config.sh

But it should do only if the word "telnetd" is not present in the file (so only the 1st time).
Strange behavior...

tunerooster, 2022-10-11:
All 5 cams have distinct client id's as showing in the mqtt view of the web
interface.  This one is "shed".  So apparently that's not it.  Please
advise further...

Thanks!

roleoroleo, 2022-10-14:
I don't know the reason of this problem, but you can fix it removing the wrong lines from init.sh.
Run vi editor inside the cam and remove the unneede lines.
Only one section
```
# Running telnetd
/usr/sbin/telnetd &

if [ -f /tmp/sd/lower_half_init.sh ];then
    source /tmp/sd/lower_half_init.sh
elif [ -f /home/app/lower_half_init.sh ];then
    source /home/app/lower_half_init.sh
else
    source /backup/lower_half_init.sh
fi
```
must be present.