roleoroleo/yi-hack-Allwinner-v2 #431: RTSP and ONVIF stop working (escam PT202)

orion131313, 2022-05-03:
Hi,
I installed the firmware on my pt202 escam.
RTSP: ON | RTSP alternative server: ON
ONVIF: ON | ONVIF WS-Discovery: OFF

The first time I tested the camera in vlcplayer or in a camera software it was working. The next day when I tested it it didn't work. I couldn't connect to the camera. I restarted the camera and it was working again, however, the next day it blocked the connection again. It only works when I restart every day...

orion131313, 2022-05-05:
I only use it sometimes to see the outside door of my house. It's not full-time.
I use windows vlcplayer or Security Eye ONVIF software. On smartphone I use the vlcplayer app or RTSP player app.
I used the RTSP alternative server because otherwise I couldn't get an image, it was always a black screen.

YoykoKoszko, 2022-05-29:
I have the same problem on my cameras the one I upgraded to 0.2.2 RTSP stopped working then I done the same for  the 2nd one and ups RTSP stopped working too.

unintended-consequences, 2022-06-08:
Same issue. RTSP working on 0.2.0. Upgraded using http button to 0.2.2. -> original RTSP no long working. But alternative server seems to be ok.

orion131313, 2022-06-09:
> Could you send me a copy of the file /dev/shm/fshare_frame_buf?
[fshare_frame_buf.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8868780/fshare_frame_buf.zip)

unintended-consequences, 2022-06-09:
> Could you send me a copy of the file /dev/shm/fshare_frame_buf?
[fshare_frame_buf.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8869031/fshare_frame_buf.zip)

roleoroleo, 2022-06-09:
The file matches and the code is ok, I don't know why it doesn't work.

orion131313, 2022-06-10:
> The file matches and the code is ok, I don't know why it doesn't work.
[fshare_frame_buf.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8878887/fshare_frame_buf.zip)

Yesterday when I sent the file the RTSP/ONVIF was working. Today I can no longer access. I send the file again (I don't know if there was any change).

closed 2023-01-07 by stale[bot]