roleoroleo/yi-hack-Allwinner-v2 #432: The h30ga model 0.2.2: RSTP stopped working

luciust, 2022-05-05:
Tested: default and alternative RTSP servers

The RSTP links are dead - no connection via VLC or other tools

Last working FW: 0.1.9

roleoroleo, 2022-05-06:
Please send me info about the processes.
Check if the processes are running:
`ps | grep -i rtsp`
Check if the port is open:
`netstat -an | grep 554`
Try the last command even while you are playing the stream.

roleoroleo, 2022-05-06:
Please, try the last commit:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/commit/bf96b2413a9512e969afd5a43cc3113c7ddc48e1

luciust, 2022-05-06:
Aw man, sorry, I forgot there's ssh console! So, netstat is not present but the process seems to be working when:
- I've configured both TWO streams
- enabled 'alternative' rtsp server

ps shows now:

`rtsp_server_yi -m h30ga -r both -a no -p 554`

and camera works back with Shinobi/VLC. No need to upgrade to latest commit!

closed 2022-05-06 by luciust