roleoroleo/yi-hack-Allwinner-v2 #433: $PATH in /etc/profile not containing yi-hack bin folders

SebLz, 2022-05-05:
Hello, as I'm using this great hack on my h30ga cam, I noticed a difference with my other cam using the [mstar](https://github.com/roleoroleo/yi-hack-MStar) hack:

In in /etc/profile of Allwinner-v2 we have:
`PATH="/usr/bin:/usr/sbin:/bin:/sbin:/home/base/tools:/home/app/localbin:/home/base"`
while in mstar you have by default the complete env setup:
`PATH="/usr/bin:/usr/sbin:/bin:/sbin:/home/base/tools:/home/app/localbin:/home/base:/home/yi-hack/bin:/home/yi-hack/sbin:/tmp/sd/yi-hack/bin:/tmp/sd/yi-hack/sbin"`

Is there any reason why the $PATH var is not setup directly when we ssh to the camera like with the mstar hack ?

roleoroleo, 2022-05-07:
Yes. The rootfs is read-only.
To change the profile I should rewrite the whole rootfs partition and it's a dangerous operation.
I created env.sh script in /tmp/sd/yi-hack/script for this purpose.
You can source it when you enter in a new console.

SebLz, 2022-05-07:
Thanks for the explanations. I was interested to have the env set up automatically for tools like WinSCP but I found a workaround.
/etc/profile can be edited but is replaced each time the camera reboots. SO I've just added this
```
echo 'PATH="/usr/bin:/usr/sbin:/bin:/sbin:/home/base/tools:/home/app/localbin:/home/base:/home/yi-hack/bin:/home/yi-hack/sbin:/tmp/sd/yi-hack/bin:/tmp/sd/yi-hack/sbin"' >> /etc/profile
echo 'export PATH' >> /etc/profile
```
in startup.sh and it seems to do the job.

closed 2022-05-07 by SebLz

roleoroleo, 2022-05-10:
Correct!
When the cam starts it copies the /etc folder in /tmp, so it becomes writable.
But this trick works only it the "Disable Cloud" option is on.