roleoroleo/yi-hack-Allwinner-v2 #437: Y28GA - No backup

noman1228, 2022-05-13:
Anyone have a backup of mtdblock4 for the Y28GA ? 
IDK where to start here...

[  578.436220] Restarting Linux version 4.9.118 (gaozhenyu@dell-PowerEdge-R740) (gcc version 6.4.1 (OpenWrt/Linaro GCC 6.4-2017.11 2017-11) ) #11 PREEMPT Wed Nov 11 02:27:00 UTC 2020
[  578.436220]
[126]HELLO! BOOT0 is starting Nov 14 2019 20:01:31!
[131]BOOT0 commit : 81c18ee
[133]board init start
[136]set pll start
[138]set pll end
[140][pmu]: bus read error
[142]board init ok
[144]chip id check OK
[146]DRAM BOOT DRIVE INFO: V0.41
[150]DRAM CLK = 528 MHz
[152]DRAM Type = 2 (2:DDR2,3:DDR3)
[155]DRAMC read ODT  off.
[158]DRAM ODT off.
[160]DRAM SIZE =64 M
[167]DRAM simple test OK.
[169]rtc standby flag is 0x0, super standby flag is 0x0
[175]dram size =64
[177]spinor id is: c8 40 17, read cmd: 03
[181]Succeed in reading toc file head.
[185]The size of toc is 4c000.
[238]Entry_name        = optee
[241]Entry_name        = u-boot
[245]Entry_name        = dtb
[248]Jump to secend Boot.
MESSAGE: [0x0] TEE-CORE: OP-TEE version: sun8iw19p1_v0.6.0-12-g97f2688 #1 2019年 11月 08日 星期五 08:26:51 UTC arm
ERROR:   [0x0] TEE-CORE:platform_standby_fdt_parse:126: no pmu node
ERROR:   [0x0] TEE-CORE:sunxi_twi_parse_from_dt:84: no pmu node

U-Boot 2018.05-00015-g8fe585e (Nov 11 2020 - 10:25:23 +0800) Allwinner Technology

[00.293]DRAM:  64 MiB
[00.296]Relocation Offset is: f9f94000
[00.307]secure enable bit: 0
[00.310]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=132Mhz
[00.316]gic: sec monitor mode
[00.318]flash init start
[00.321]workmode = 0,storage type = 3
SF: Detected gd25q64 with page size 256 Bytes, erase size 4 KiB, total 8 MiB
[00.337]sunxi flash init ok
[00.340]Loading Environment from SUNXI_FLASH... start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x3e00, len: 0x100
OK
[00.401]try sprite_led_gpio config
xxxxxxxxxxxxxxlijun uboot xiaoyiledinit
[00.409]sprite_led_gpio start
[00.412](weak)update dtb dram start
[00.416]update dtb dram  end
[00.419]update dts
root_partition is rootfs
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
set root to /dev/mtdblock2
[00.481]update part info
[00.484]update bootcmd
mmc driver ver uboot2018:2019-8-15 16:34:00
get mem for descripter OK !
[00.496]get sdc_wipe fail.
[00.498]get sdc0 sdc_erase fail.
[00.501]get sdc0 sdc_boot fail.
[00.504]get sdc0 sdc_odly_50M fail.
[00.507]get sdc0 sdc_sdly_50M fail.
[00.511]get sdc0 sdc_odly_50M_ddr fail.
[00.514]get sdc0 sdc_sdly_50M_ddr fail.
[00.518]get sdc0 sdc_freq fail.
[00.521]get sdc0 sdc_b0p fail.
[00.523]get card0_boot_para:sdc_ex_dly_used fail
[00.529]get card-pwr-gpios handler:1119307360
[00.532]get card0_boot_para:time_pwroff:200ms
[00.538]Using default timing para
[00.741]init mmc 0 clock and io
devnum 0, prv 43fbd43c, bdesc 42b74844
SUNXI SD/MMC: 0
[00.748]==================== work mode: 0 0, sample_mode:0
[00.754]=============== start mmc_init_boot...
card_caps:0x3000000a
host_caps:0x3000003f
[00.854]unable to read ssr
[00.857]unable to read ssr
[00.861]the mode SD Legacy (freq : 25 MHz)
** Unrecognized filesystem type **
check_file [/one_y28ga.bin] exsit
check_file [/one_y28ga.bin] exsit
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
part_name:boot start_offset:0x20 part_size:1900544
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
part_name:rootfs start_offset:0xea0 part_size:1179648
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
part_name:home start_offset:0x17a0 part_size:3407872
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
part_name:backup start_offset:0x31a0 part_size:1245184
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
part_name:env start_offset:0x3b20 part_size:131072
Nothing updated, start normal boot
Hit any key to stop autoboot:  0
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
[02.002]partinfo: name boot, start 0x20, size 0xe80
start: 0x300, len: 0x40
start: 0x340, len: 0xe12
[02.451]android.hardware = sun8iw19p1
## Booting kernel from Legacy Image at 45000000 ...
   Image Name:   ARM OpenWrt Linux-4.9.118
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    1876872 Bytes = 1.8 MiB
   Load Address: 40008000
   Entry Point:  40008000
[02.518]Starting kernel ...

** 2 printk messages dropped ** [    0.000000] ion_mem_reserve: err format for ion reserve list!
[    0.097757] ion_heap_create: Invalid heap type 4
[    0.103983] sunxi_i2c_probe()2209 - [i2c1] warning: failed to get regulator id
[    0.127618] uart uart0: get regulator failed
[    0.158186] sunxi-wlan wlan: request pincrtl handle for device [wlan] failed
[    0.166465] sunxi-wlan wlan: get gpio chip_en failed
[    0.190022] mtd: partition "UDISK" is out of reach -- disabled
[    0.282785] VE: get debugfs_mpp_root is NULL, please check mpp
[    0.282785]
[    0.556185] mmc:failed to get gpios
[    0.604536] sunxi-mmc sdc1: smc 1 p1 err, cmd 52, RTO !!
[    0.611411] sunxi-mmc sdc1: smc 1 p1 err, cmd 52, RTO !!
[    0.621869] sunxi-mmc sdc1: smc 1 p1 err, cmd 8, RTO !!
[    0.628613] sunxi-mmc sdc1: smc 1 p1 err, cmd 5, RTO !!
[    0.635395] sunxi-mmc sdc1: smc 1 p1 err, cmd 5, RTO !!
[    0.642171] sunxi-mmc sdc1: smc 1 p1 err, cmd 5, RTO !!
[    0.648931] sunxi-mmc sdc1: smc 1 p1 err, cmd 5, RTO !!
[    0.655736] sunxi-mmc sdc1: smc 1 p1 err, cmd 55, RTO !!
[    0.662635] sunxi-mmc sdc1: smc 1 p1 err, cmd 55, RTO !!
[    0.669255] --->sunxi_daudio_dev_probe<--- strat.
[    0.674604] sunxi-mmc sdc1: smc 1 p1 err, cmd 55, RTO !!
[    0.680718] sunxi-mmc sdc1: smc 1 p1 err, cmd 55, RTO !!
[    0.686907] sunxi-mmc sdc1: smc 1 p1 err, cmd 1, RTO !!
[    0.694214] --->sunxi_daudio_dev_probe<--- end.
starting pid 536, tty '': '/etc/init.d/rcS boot'
[RCS]: /etc/init.d/S01udev
Populating /dev using udev: [    1.316410] udevd[541]: could not create /run/udev: No such file or directory
done
[RCS]: /etc/init.d/S02app
/backup/init.sh not exist!
[RCS]: /etc/init.d/S20urandom
read-only file system detected...done
starting pid 552, tty '/dev/console': '-/bin/sh'

BusyBox v1.27.2 () built-in shell (ash)

            __    __   _
            \ \  / /  | |
             \ \/ /   | |
              \  /    | |
              / /     | |
             /_/      |_|
 ------------------------------------
 Welcome to XiaoYi Linux.
 ------------------------------------

root@(none):/#

roleoroleo, 2022-05-15:
[mtdblock4.bin.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8694607/mtdblock4.bin.gz)

noman1228, 2022-05-15:
Up and running! Thank you again sir!

noman1228, 2022-05-15:
Maybe not. 

1) Flashed and booted correctly
2) added with app
3) got image for about 60 seconds
4) Bootloop

[60]HELLO! BOOT0 is starting Nov 14 2019 20:01:31!
[65]BOOT0 commit : 81c18ee
[68]board init start
[70]set pll start
[72]set pll end
[74][pmu]: bus read error
[77]board init ok
[78]chip id check OK
[80]DRAM BOOT DRIVE INFO: V0.41
[83]DRAM CLK = 528 MHz
[86]DRAM Type = 2 (2:DDR2,3:DDR3)
[89]DRAMC read ODT  off.
[91]DRAM ODT off.
[94]DRAM SIZE =64 M
[100]DRAM simple test OK.
[103]rtc standby flag is 0x0, super standby flag is 0x0
[108]dram size =64
[111]spinor id is: c8 40 17, read cmd: 03
[115]Succeed in reading toc file head.
[118]The size of toc is 4c000.
[171]Entry_name        = optee
[174]Entry_name        = u-boot
[178]Entry_name        = dtb
[182]Jump to secend Boot.
MESSAGE: [0x0] TEE-CORE: OP-TEE version: sun8iw19p1_v0.6.0-12-g97f2688 #1 2019年 11月 08日 星期五 08:26:51 UTC arm
ERROR:   [0x0] TEE-CORE:platform_standby_fdt_parse:126: no pmu node
ERROR:   [0x0] TEE-CORE:sunxi_twi_parse_from_dt:84: no pmu node

U-Boot 2018.05-00015-g8fe585e (Nov 11 2020 - 10:25:23 +0800) Allwinner Technology

[00.227]DRAM:  64 MiB
[00.230]Relocation Offset is: f9f94000
[00.241]secure enable bit: 0
[00.243]CPU=816 MHz,PLL6=600 Mhz,AHB=200 Mhz, APB1=100Mhz  MBus=132Mhz
[00.250]gic: sec monitor mode
[00.252]flash init start
[00.255]workmode = 0,storage type = 3
SF: Detected gd25q64 with page size 256 Bytes, erase size 4 KiB, total 8 MiB
[00.271]sunxi flash init ok
[00.274]Loading Environment from SUNXI_FLASH... start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x2e0, len: 0x1
start: 0x2e1, len: 0x1
start: 0x2e2, len: 0x2
start: 0x3e00, len: 0x100
OK
[00.335]try sprite_led_gpio config
xxxxxxxxxxxxxxlijun uboot xiaoyiledinit
[00.343]sprite_led_gpio start
[00.346](weak)update dtb dram start
[00.350]update dtb dram  end
[00.352]update dts
root_partition is rootfs
[... 435 lines cut ...]

watch_process.c(get_watch_info-134)[00:07:31.346]:process=rmm;cmd=reboot;

watch_process.c(get_watch_info-134)[00:07:31.346]:process=p2p_tnp;cmd=cd /home/app;./p2p_tnp &;

watch_process.c(get_watch_info-134)[00:07:31.347]:process=mp4record;cmd=cd /home/app;./mp4record &;

watch_process.c(get_watch_info-134)[00:07:31.347]:process=oss;cmd=cd /home/app;./oss &;

v429_mpp.c(track_process-3088)[00:07:31.360]:ptz_isruning ==1 so return

v429_mpp.c(track_process-3088)[00:07:31.556]:ptz_isruning ==1 so return

v429_mpp.c(track_process-3088)[00:07:31.751]:ptz_isruning ==1 so return

v429_mpp.c(track_process-3088)[00:07:31.947]:ptz_isruning ==1 so return

[   11.335590] sunxi-mmc sdc0: smc 0 p0 err, cmd 18, RD RTO !!
[   11.363131] sunxi-mmc sdc0: smc 0 p0 err, cmd 17, RD RTO !!
[   11.369496] sunxi-mmc sdc0: send  manual stop command failed
[   11.376990] sunxi-mmc sdc0: smc 0 p0 err, cmd 17, RD DCE !!
[   11.383362] sunxi-mmc sdc0: send  manual stop command failed
[   11.390097] sunxi-mmc sdc0: smc 0 p0 err, cmd 17, RD RTO !!
[   11.398076] sunxi-mmc sdc0: smc 0 p0 err, cmd 18, RD EBE !!
[   11.405530] sunxi-mmc sdc0: smc 0 p0 err, cmd 18, RD DCE !!

noman1228, 2022-05-15:
Threw it in the trash and bought Amcrest. Thanks for all of your help Roleo. You're a fantastic code wizard.

closed 2022-05-15 by noman1228

roleoroleo, 2022-05-16:
Did you apply the hack after the restore procedure?