roleoroleo/yi-hack-Allwinner-v2 #439: Escam PT202

macedpo, 2022-05-22:
I have a PT202 with installed Hack B091QP .
Is there a possibility to provide a HTTP Port such as 8203.
I would like to use the PT202 with my DVR that uses a NETIP ----> 192.168.0.129:8203
On my DVR the Channel status is: Mainstream/Sub-stream --->  1080P/UNKNOWN
The Sub-stream is not detected on my DVR however the RSTP stream is set to "BOTH"
On all my other IP ONVIF/RSTP cams  i see both streams.
Any thoughts ?
Many thanks.

roleoroleo, 2022-05-23:
> Is there a possibility to provide a HTTP Port such as 82xx.

You can change the TCP ports in the configuration web page. But I don't understand how this will help you.

IMHO onvif is not a good standard and there are many compatibility problems.
I don't know what your DVR expects and onvif_srvd contains only a partial implementation of the whole protocol.

macedpo, 2022-05-23:
Hi Roleo,
Thanks for your reply.
On the web page i see indeed RTSP port, Onvif port & HTTPD port but no HTTP port.
My problem is that motion detection doesn't work if i use ONVIF on my DVR (cheap one).
I have also 2 Escam QF218 (D02 & D03 on the picture) and i must configure them as NETIP with HTTP port 82xx and then motion detection works fine. Dont ask my why...
Any idea why the sub stream is not recognized (see picture) ?

roleoroleo, 2022-05-23:
> On the web page i see indeed RTSP port, Onvif port & HTTPD port but no HTTP port.

HTTPD is the daemon that provides HTTP.

> My problem is that motion detection doesn't work if i use ONVIF on my DVR (cheap one).

Events are not implemented in this onvif daemon.

> I have also 2 Escam QF218 (D02 & D03 on the picture) and i must configure them as NETIP with HTTP port 82xx and then motion detection works fine. Dont ask my why...

This cam doesn't implement NETIP protocol, you can't use it.

> Any idea why the sub stream is not recognized (see picture) ?

No, sorry.

macedpo, 2022-05-23:
Thanks Roleo for your answer.
Great hack anyway !

macedpo, 2022-05-24:
Replaced onvif_srvd but no change. Sub-stream unknown on my NVR and black picture on the mosaic. However if i click on the black picture in the mosaic i get a full screen HD image (as before).

macedpo, 2022-05-24:
Roleo,
I've activated the 2 sliders RTSP server + RTSP altenative server and now i get in the channel status ---> mainstream/substream ---> 1080P/QVGA and see now a perfect image on the mosaic. I'm sure i did that before but with no + results.
So it must be your new onvif_srvd file who did the job !
1000 times thanks !!

closed 2023-01-08 by stale[bot]