roleoroleo/yi-hack-Allwinner-v2 #441: xiaomi imilab ec2

tv4you2016, 2022-05-26:
hello, what is the possibility of this hack working on a xiaomi imilab ec2 (CMSXJ11AG)?

roleoroleo, 2022-05-26:
I don't know.
Please post the firmware version you read in the app.
And take some picture of the internal board.

tv4you2016, 2022-05-29:
The gateway unit is using a Mediatek Mt7628NN SOC, running a customized version of OpenWrt (version 14, if I recall).
It has 64MB of RAM and 8MB of flash (as far as I could tell).

There is a serial (UART) console connector, close to the usb power port, connect at speed 57600 bps.

roleoroleo, 2022-05-29:
This hack can't work with your cam.
It needs a new hack.

tv4you2016, 2022-05-30:
what do you need to do it?? Or can you give me some tips
where to start??

roleoroleo, 2022-05-31:
Sorry but I don't own this cam and I can't create a hack at the moment.

closed 2022-06-10 by tv4you2016