roleoroleo/yi-hack-Allwinner-v2 #442: ssh with authorized_keys (no password) stopped working

tunerooster, 2022-05-29:
Something changed.  I'm not sure when, but it must have been with the 2.2 update...

```ssh root@camera``` without a password no longer works (on all cameras).

The authorized_keys file in the ~/.ssh directory is correct (as I have been using it for a long time).
But trying to ssh into the camera without a password stopped working.  I have of course re-authorized many time and tried everything else I can thing of with no luck.

I use passwordless ssh in several scripts which are now all failing.

I tried setting the ssh password to null in the web menu, bit it still asks for a password.

Did something in dropbear change?  Is there a way to fix it?

Thanks as always!!!

roleoroleo, 2022-05-31:
I never used login without password.
But I tested it now with the last version and it seems working properly.
Are you using the correct path?
`/tmp/sd/yi-hack/.ssh/`

roleoroleo, 2022-05-31:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/SSH:-login-with-keys

tunerooster, 2022-05-31:
Yes, I had been using it for years and just recently discovered it was no
longer working.  If it is working for you, I will have to dig even deeper
into it and get back to you.

tunerooster, 2022-05-31:
I noticed in the instructions you posted, it says "chmod 0600".  When I do
that (as root  of course) the permissions do not change.  In fact, all
permissions on all files I have checked are " -rwxr-xr-x" and cannot be
changed.  Is this significant?  Are you seeing the same thing?

tunerooster, 2022-06-01:
I am running (from ssh -V):
```
OpenSSH_9.0p1, OpenSSL 1.1.1o  3 May 2022
```
What are you running?  There was/is a security vulnerability found in id_rsa signatures and id_rsa is being depricated.  I suspect this is the problem:
```
https://confluence.atlassian.com/bitbucketserverkb/ssh-rsa-key-rejected-with-message-no-mutual-signature-algorithm-1026057701.html
```
I may have do downgrade OpenSSH (I updated it in the last month).  According to the link above, there is a workaround (on the OpenSSH server end):
```
PubkeyAcceptedKeyTypes +ssh-rsa
```
But dropbear does not offer this.  It appears here is a 2022 version of dropbear which supports ED25519 (which is the recommended key algorithm).  I tried ```ssh-keygen ed25519`` but apparently dropbear 2018 does not support it.

UPDATE:

I subsequently tried ECDSA and it worked!

So I'm good to go, but if you do upgrade dropbear to 2022, I'll be happy to test ed25519.  You might want to add/change the posted instructions to reflect this (for those with the latest OpenSSH).

Best regards!

roleoroleo, 2022-06-01:
> I noticed in the instructions you posted, it says "chmod 0600". When I do that (as root of course) the permissions do not change. In fact, all permissions on all files I have checked are " -rwxr-xr-x" and cannot be changed. Is this significant? Are you seeing the same thing?

You are right, in the Allwinner-v2 platform, the hack runs from the sd card, a fat32 filesystem.
So chmod command has no effect.

> So I'm good to go, but if you do upgrade dropbear to 2022, I'll be happy to test ed25519. You might want to add/change the posted instructions to reflect this (for those with the latest OpenSSH).

I will add this update to the todo list.

roleoroleo, 2022-06-03:
The good news is that I updated dropbear and ed2559 works.
The bad news is that I tried again to authorize logins with keys and I'am not able to make it working.
Something is wrong in my previous posts.
Which file/path are you using for authorize_keys?

```
/tmp/sd/yi-hack/.ssh
/root/.ssh
...
```

tunerooster, 2022-06-03:
I'm using:  /tmp/sd/yi-hack/.ssh

ls -l /tmp/sd/yi-hack/.ssh
-rwxr-xr-x    1 root     root          1759 Jun  1 06:46 authorized_keys
-rwxr-xr-x    1 root     root             0 Dec  4 09:46 known_hosts

ls /root/.ssh
ls: /root/.ssh: No such file or directory

Have you tried "sh -vvvv".  It may offer clues...

tunerooster, 2022-06-03:
If you want to send me the new dropbear, I can try it here...

roleoroleo, 2022-06-04:
I will send you asap.

roleoroleo, 2022-06-04:
Here it is:
[dropbearmulti.gz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8837711/dropbearmulti.gz)

tunerooster, 2022-06-05:
It works here!  Output from the new ```/tmp/sd/yi-hack/bin/dropbearmulti```:
```
Dropbear SSH multi-purpose v2022.82
Make a symlink pointing at this binary with one of the
following names or run 'dropbearmulti <command>'.
'dropbear' - the Dropbear server
'dbclient' or 'ssh' - the Dropbear client
'dropbearkey' - the key generator
'dropbearconvert' - the key converter
'scp' - secure copy
```
The relevant debug output from ```ssh -v root@camera``` is:
```
debug1: identity file /home/rwk/.ssh/id_ed25519 type 3
debug1: identity file /home/rwk/.ssh/id_ed25519-cert type -1
debug1: identity file /home/rwk/.ssh/id_ed25519_sk type -1
debug1: identity file /home/rwk/.ssh/id_ed25519_sk-cert type -1
debug1: Will attempt key: /home/rwk/.ssh/id_ed25519 ED25519 SHA256:VXeDL5JL/A8x7sJSD0PGVy05eCthkOkwrj3T4ppPYUc agent
debug1: Will attempt key: /home/rwk/.ssh/id_ed25519_sk 
debug1: kex_input_ext_info: server-sig-algs=<ssh-ed25519,sk-ssh-ed25519@openssh.com,ecdsa-sha2-nistp256,ecdsa-sha2-nistp384,ecdsa-sha2-nistp521,sk-ecdsa-sha2-nistp256@openssh.com,rsa-sha2-256,ssh-rsa,ssh-dss>
debug1: Offering public key: /home/rwk/.ssh/id_ed25519 ED25519 SHA256:VXeDL5JL/A8x7sJSD0PGVy05eCthkOkwrj3T4ppPYUc agent
debug1: Server accepts key: /home/rwk/.ssh/id_ed25519 ED25519 SHA256:VXeDL5JL/A8x7sJSD0PGVy05eCthkOkwrj3T4ppPYUc agent
```
I forgot to answer your question...
* I *have* changed the root password.  It is not null.

I did the following on the client side:
```
ssh-keygen -t ed25519
#  Copy contents of:  ~/.ssh/id_ed25519.pub to a new completely empty: /tmp/sd/yi-hack/.ssh/authorized_keys
#  I.e., there are no id_rsa or any other authorized keys in the authorized_keys file on the camera. 
ssh -v root@camera
```
And it comes right up...  Should I install this new dropbearmulti on all my cameras now, or wait for an update?

roleoroleo, 2022-06-06:
> And it comes right up... Should I install this new dropbearmulti on all my cameras now, or wait for an update?

As you want.
I already committed this change. It will be included in the next release.

tunerooster, 2022-06-06:
I'll wait for the next release...  (less work).

:thumbsup: