roleoroleo/yi-hack-Allwinner-v2 #444: RFUS - YI Home 1080p - Cannot dump firmware or flash the camera

BabaJaga2600, 2022-05-30:
I have a few older YI Home's I have been able to flash no issues.  This new model seems to resist any interaction.  The chip set is labeled Y291GA_MB_V3.1 2021/11/22.  I've tried y211ga_0.2.2 y29ga_0.2.2.  Then I wanted to dump the firmware to get the version.  It seems to completely ignore the files on the SD and reboots to normal operation.  I've tried a few models of SD cards.  Any advice is much appreciated.

roleoroleo, 2022-08-17:
It should be a y291ga.
Try another sd card, backup operation must work.

Al3eRtO, 2022-08-29:
> Ho alcuni YI Home più vecchi che sono stato in grado di eseguire il flashing senza problemi. Questo nuovo modello sembra resistere a qualsiasi interazione. Il chipset è etichettato come Y291GA_MB_V3.1 22/11/2021. Ho provato y211ga_0.2.2 y29ga_0.2.2. Quindi volevo scaricare il firmware per ottenere la versione. Sembra ignorare completamente i file sulla SD e si riavvia al normale funzionamento. Ho provato alcuni modelli di schede SD. Qualsiasi consiglio è molto apprezzato. Il codice QR sul retro è RFUSY623VG

Hai risolto? Io ho la YI RFUSY625HE

roleoroleo, 2022-09-01:
Please, check the fw version inside the app.

BabaJaga2600, 2022-09-05:
Hi @roleoroleo and @Al3eRtO  
I'll try another card, I've tried with a two brands but I try again with another one.  Both cards had the same issue.  And I'm using the same procedure I used with the older cameras.  So far this camera just will not flash.
_Proverò un'altra carta, ho provato con una delle due marche ma proverò di nuovo con un'altra. Entrambe le carte hanno avuto lo stesso problema. E sto usando la stessa procedura che ho usato con le fotocamere più vecchie. Finora questa fotocamera non si aggiornerà_

roleoroleo, 2022-09-06:
I can confirm. It's a y291ga.
The reason it doesn't work could be:
- wrong structure of files and dirs
- unsupported sd card
- incorrect format (only fat32 filesystem is supported)
- hardware failure

Check if you properly copied the files into the sd card.
Try to format the sd card with a different method: linux, windows, ...

BabaJaga2600, 2022-09-10:
@roleoroleo 
I reformatted the card with Windows and deselected "Quick Format"
Then used 7zip to unzip the newest y291ga version to the root of the drive.
The hack is working!!!

Thank you for making this awesome hack!
Thank you for your support!

closed 2022-09-10 by BabaJaga2600