roleoroleo/yi-hack-Allwinner-v2 #448: Yi outdoor floodlight bricked

Sp4rky001, 2022-06-05:
Hi everyone

I wonder if anyone can point me in the right direction. I have the Yiiot floodlight – a product that doesn’t see to exist, except on Amazon (it’s not even on their website).

My firmware was 9.41 and I saw that others tried the b091qp firmware with another camera successfully. I popped the SD card in and now the camera isn’t booting.  I do have the backup of my original firmware.  Is there any way to use that to create a recovery image for my camera?

Also, from that firmware backup, is there a way to create a custom firmware?

roleoroleo, 2022-06-06:
Share your backup.
I can check it it's possible.

Sp4rky001, 2022-06-06:
Thank you!  Here is my backup
[backup.zip](https://github.com/roleoroleo/yi-hack-Allwinner-v2/files/8846171/backup.zip)

roleoroleo, 2022-06-07:
Your cam is very similar to this one:
https://github.com/roleoroleo/yi-hack_ha_integration/issues/84

And b091qp should work properly.
If you want to restore the original partition check the wiki:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Unbrick-the-cam

Sp4rky001, 2022-06-07:
thank you for your help.  Before I posted my first message, I tried to debrick using the software method using the b091qp_oscam folder but I will try the blitzwolf one.  It's tricky for me because it is installed on my house.  I have to run back and forth to my back yard.  It might take me a few days to try.

Sp4rky001, 2022-06-07:
I'm sorry.  Do you know if I should rename the generated file when I copy it to the SD card?

roleoroleo, 2022-06-08:
There are no difference between b091qp_blitzwolf and b091qp_oscam.
You can use whatever you want.

The name must be backup_b091qp

Sp4rky001, 2022-06-10:
Yes!  I'm happy to report that the unbrick process worked, but only with ./build.sh factory.  Trying the hack again will brick it.

I think that maybe there's a chance that my backup isn't clean.  This backup may be from one where I've already tried Allwinner v1.

You wouldn't know how to restore the partition to completely factory, would you?

roleoroleo, 2022-06-11:
> I think that maybe there's a chance that my backup isn't clean. This backup may be from one where I've already tried Allwinner v1.

Your backup is ok.

> You wouldn't know how to restore the partition to completely factory, would you? Specifically, if I hold the reset button to do a factory reset, will it also clear out any previous hacks?

If you hold the button the cam will reset the configuration but will not remove the hack.

> An update: After unbricking, I performed a factory reset and set up the wifi again. Then I tried the hack for b091qp again. It bricked again.

How did you apply the hack? Using the unbrick procedure `./build.sh hacked` or the standard hacking procedure?

roleoroleo, 2022-06-11:
> Yes! I'm happy to report that the unbrick process worked, but only with ./build.sh factory. Unfortunately, Trying the hack again will brick it.

Maybe the build.sh exits with an error.
For example if mtd-utils are not installed.
Check the size of the file backup_b091qp.gz at the end of the command.

Sp4rky001, 2022-06-12:
> Your backup is ok.
>
>
> If you hold the button the cam will reset the configuration but will not remove the hack.
>
>
> How did you apply the hack? Using the unbrick procedure `./build.sh hacked` or the [...]

I tried with both "./build.sh hacked" as well as b091qp_0.2.2.tgz from the repository's releases.  Both show the same behavior:
1.  Plug in the power and the flood lights come on, the motion sensing also still works.
2.  I leave the camera alone for at least 10 minutes.  During this time, I check the yiiot app and the camera doesn't show up.  I try to reach the snapshot URL of the camera but the page never shows.

Fortunately, I've been able to restore using ./build.sh factory.

The size of backup_091qp.gz:
-  factory: 1240546
-  hacked: 2491

It seem that you are right with build.sh failing.  Looking more closely, I ran `sudo ./build.sh hacked` and got this:

2688+0 records in
2688+0 records out
1376256 bytes (1.4 MB, 1.3 MiB) copied, 0.00703039 s, 196 MB/s
Successfully mounted b091qp_blitzwolf/mtdblock4.bin on b091qp_blitzwolf/mnt
### Updating b091qp_blitzwolf/backup/init.sh
cp: cannot stat 'b091qp_blitzwolf/backup/init.sh': No such file or directory
sed: can't read b091qp_blitzwolf/backup/tmp_init.sh: No such file or directory
sed: can't read b091qp_blitzwolf/backup/tmp_init.sh: No such file or directory
Image Name:   xiaoyi-backup
Created:      Sat Jun 11 20:02:37 2022
Image Type:   ARM Linux Filesystem Image (uncompressed)
Data Size:    1376256 Bytes = 1344.00 KiB = 1.31 MiB
Load Address: 00000000
Entry Point:  00000000

Then I tried running each line manually starting from line 46.
mount.jffs2 gave me this:

2688+0 records in
2688+0 records out
1376256 bytes (1.4 MB, 1.3 MiB) copied, 0.00926016 s, 149 MB/s
Successfully mounted b091qp_blitzwolf/mtdblock4.bin on b091qp_blitzwolf/mnt

but the mnt folder was empty.
I'm not sure how to mount the bin file properly

roleoroleo, 2022-06-13:
Try to pull the repo again, I updated the unbrick part.

Sp4rky001, 2022-06-13:
> Try to pull the repo again, I updated the unbrick part.

Hi, I deleted my local repo and performed a clone again.  ./build.sh hacked is still giving the same error for mtdblock4.bin.  I checked the git commits and saw that only the readme was updated and the folder was renamed.  Did you have a change to build.sh?

Should I run ./build.sh factory with all three files (mtdblock2.bin, mtdblock3.bin, mtdblock4.bin) and try the hack again?

An update on my device:  One of the floodlights stopped working on my device and I contacted the seller.  They offered to refund 50% when I order a second unit.  So I will be able to get a clean factory backup and I'll also be able to try the hack on a second device.

Once I get that second unit, I'll update again.  First thing I'll do is to create a backup.

roleoroleo, 2022-06-14:
> Should I run ./build.sh factory with all three files (mtdblock2.bin, mtdblock3.bin, mtdblock4.bin), do a restore, then try the hack again using the scripts?

No, mtdblock4.bin should be enough.

Try this comment some lines in the script:
```
#!/bin/bash

update_init()
{
    ### Update /backup/init.sh with a more friendly one
    echo "### Updating $DIR/backup/init.sh"
    cp $DIR/backup/init.sh $DIR/backup/tmp_init.sh
    sed -n '1{$!N;$!N;$!N;$!N};$!N;s@\nif \[ \-f \/home\/app\/lower_half_init.sh \];then\n    source \/home\/app\/lower_half_init.sh\nelse\n    source \/backup\/lower_half_init.sh\nfi@@;P;D' -i $DIR/backup/tmp_init.sh
    sed -n '1{$!N;$!N;$!N;$!N};$!N;s@\nif \[ \-f \/home\/app\/lower_half_init.sh \];then\n\tsource \/home\/app\/lower_half_init.sh\nelse\n\tsource \/backup\/lower_half_init.sh\nfi@@;P;D' -i $DIR/backup/tmp_init.sh

    echo "# Running telnetd" >> $DIR/backup/tmp_init.sh
    echo "/usr/sbin/telnetd &" >> $DIR/backup/tmp_init.sh
    echo "" >> $DIR/backup/tmp_init.sh
    echo "if [ -f /tmp/sd/lower_half_init.sh ];then" >> $DIR/backup/tmp_init.sh
    echo "    source /tmp/sd/lower_half_init.sh" >> $DIR/backup/tmp_init.sh
    echo "elif [ -f /home/app/lower_half_init.sh ];then" >> $DIR/backup/tmp_init.sh
    echo "    source /home/app/lower_half_init.sh" >> $DIR/backup/tmp_init.sh
    echo "else" >> $DIR/backup/tmp_init.sh
    echo "    source /backup/lower_half_init.sh" >> $DIR/backup/tmp_init.sh
    echo "fi" >> $DIR/backup/tmp_init.sh

    cp $DIR/backup/tmp_init.sh $DIR/backup/init.sh
    rm $DIR/backup/tmp_init.sh
}

TYPE=$1
if [ "$TYPE" != "factory" ] && [ "$TYPE" != "hacked" ]; then
    echo "Usage: $0 type"
    echo -e "\twhere type = \"factory\" or \"hacked\""
    exit
fi

for DIR in * ; do
    if [ -d $DIR ]; then
        FILENAME=`cat $DIR/filename`
        if [ -f $DIR/mtdblock2.bin ]; then
            mkimage -A arm -O linux -T filesystem -C none -a 0x0 -e 0x0 -n "xiaoyi-rootfs" -d $DIR/mtdblock2.bin $DIR/rootfs_$FILENAME
            gzip $DIR/rootfs_$FILENAME
        fi
        if [ -f $DIR/mtdblock3.bin ]; then
            mkimage -A arm -O linux -T filesystem -C none -a 0x0 -e 0x0 -n "xiaoyi-home" -d $DIR/mtdblock3.bin $DIR/home_$FILENAME
            gzip $DIR/home_$FILENAME
        fi
        if [ -f $DIR/mtdblock4.bin ]; then
            if [ "$TYPE" == "hacked" ]; then
                mkdir -p $DIR/mnt
                ./mount.jffs2 $DIR/mtdblock4.bin $DIR/mnt 4
                cp -r $DIR/mnt $DIR/backup
                umount $DIR/mnt
                rm -rf $DIR/mnt
                update_init
                if [ $FILENAME == "b091qp" ]; then
                    PAD="0x00150000"
                else
                    PAD="0x00130000"
                fi
                ./create.jffs2 $PAD $DIR/backup $DIR/mtdblock4_hacked.bin
#                rm -rf $DIR/backup
            else
                cp $DIR/mtdblock4.bin $DIR/mtdblock4_hacked.bin
            fi
            mkimage -A arm -O linux -T filesystem -C none -a 0x0 -e 0x0 -n "xiaoyi-backup" -d $DIR/mtdblock4_hacked.bin $DIR/backup_$FILENAME
            rm -f $DIR/mtdblock4_hacked.bin
            gzip $DIR/backup_$FILENAME
        fi
    fi
done
```

And check the content of the folder "backup".

Sp4rky001, 2022-06-16:
backup folder is empty when I comment out the rm -rf line.

Amazon lost my package for that replacement light so I'm still waiting before I can get a clean backup.

roleoroleo, 2022-06-16:
> backup folder is empty when I comment out the rm -rf line.

Probably the mount command fails.
You should try with another linux machine.

Sp4rky001, 2022-06-16:
some progress.  I ran ./build.sh hacked on a debian system and it got further.  It failed at:
`./create.jffs2 $PAD $DIR/backup $DIR/mtdblock4_hacked.bin`

with an error of:
`./create.jffs2: 3: ./mkfs.jffs2: Exec format error`

After some trials and errors, I modified create.jffs2 from:
`./mkfs.jffs2 -l --pad=$1 -e 4KiB -r $2 -o $3`

to:
`mkfs.jffs2 -l --pad=$1 -e 4KiB -r $2 -o $3`

The image built successfully with only this warning:
`Erase size 0x1000 too small. Increasing to 8KiB minimum`

Sp4rky001, 2022-06-16:
I have great news!  I received the new camera and I made a backup of the camera in its vanilla state.  After setting up the wifi, I made another backup.

I then tried the b091qp_0.2.2.tgz hack and I finally got a webpage at http://<ip>:8080
I was able to get a snapshot but was unable to get a camera stream.

Home Assistant was able to find the camera via ONVIF but still unable to retrieve the camera feed.

The camera seems quite unstable, though.  It would reboot either at random or when I try to do something.  Such as retrieving camera feed or browsing the web ui

Now that SSH is possible, is there some place I can check for logs?  I can send those logs over to you

roleoroleo, 2022-06-17:
Please, enable swap file in the web config.
This will improve stability.

About ha, try my custom component.

Sp4rky001, 2022-06-17:
Hi!  I'm happy to report that the replacement unit worked beautifully when swap is turned on and using your custom integration.  Thanks so much for your help!

Now for my other unit, I will continue to try and fix it.  Is it safe for me to restore the backup partition from the new camera into the old one?  I figure I'll try that then apply the hack.

Sp4rky001, 2022-06-17:
Okay, finally I think you can mark my camera down as fully supported.  I was able to run the unbrick process with the vanilla mtdblock4.bin file from the new camera.  From there, I applied the hack using [b091qp_0.2.2.tgz](https://github.com/roleoroleo/yi-hack-Allwinner-v2/releases/download/0.2.2/b091qp_0.2.2.tgz)

It took a few minutes, but eventually, the camera made a clicking sound and not long after, the startup sound rang.  I immediately enabled swap in the Web UI and now both of my cameras are working.

I want to thank you for your help over the last couple of weeks.  I truly appreciate it.  Here are the details for my camera:

Model:
L850Y-US

Description:
Flood light Camera Outdoor, Garage Light with Camera, 24/7 Recording, Colorful/Infrared Night Vision, IP66 Waterproof,180°PIR Motion Detection, Works with Alexa

closed 2022-06-20 by Sp4rky001

roleoroleo, 2022-06-25:
Thank you for your feedback.