roleoroleo/yi-hack-Allwinner-v2 #452: Flashing Yi 1080p Home RFUS whitout installing App, stuck at "Waiting to Connect"

jason0x21, 2022-06-13:
I'm trying to get a Yi 1080p Home RFUS camera up and running without bothering to install the app on my phone.  I _think_ I've managed to get the y211ga firmware installed. There's a click, and then "Welcome to Yi Home / Waiting to Connect", and when I shut it off and check the flash card, the "Factory" folder is now "Factory.done".

However, when I try to then change the WiFi credentials using the sdwifi.tar.gz file, nothing really changes, and I still get "Waiting to Connect" over and over again.  I noticed that the tar file doesn't include the "yi-hack" directory, so I tried it again with a "yi-hack" folder on the SD card and everything in the tar file _under_ that.  Still no luck. There's nothing written to the card after I remove it to tell me if it worked or not.

Am I missing something here?  Thanks!

roleoroleo, 2022-06-15:
Please, check the wiki: https://github.com/roleoroleo/yi-hack-Allwinner-v2/wiki/Change-WiFi-credentials
I changed it to make it clear.

jason0x21, 2022-06-16:
Got it working!  Thanks.  Here's what I did...

1) Downloaded the new firmware (y291ga_0.2.2.tar.gz) and extracted it onto a fresh SD card.
2) Extracted the sdwifi.tar.gz file into the "yi-hacks" directory on that same card.
3) Edited the configure_wifi.cfg with the right credentials.
4) Put the SD card in the camera and apply power.

I got the blue LED and can stream video from it.  I haven't powered it down yet, but do I need to do that and then delete the extra config files?

roleoroleo, 2022-06-17:
You can remove the 3 files related to the wifi.
You can do it without removing the sd card, for example through ssh.

closed 2023-01-07 by stale[bot]