roleoroleo/yi-hack-Allwinner-v2 #453: YI Home 1080p RFUSY628VR on FW 9.0.35 flashed with y211ga, rtsp/onvif not working

smashley3000, 2022-06-15:
So I realize the FW version in the compatibility chart says FW version should be 9.0.3**6*** and not 9.0.3**5** like I have, but figured I'd give it a go anway since I really want rtsp working on these cameras. Most everything seems to be working fine, except I get 'connection refused' when trying to hit the rtsp url, or IP:80 for onvif I get http 405 response (method not allowed). Curious if anyone else has encountered this? Is it a FW specific compatibility? Trying to find logs via ssh and/or ftp but not overly successful. I do have a second un-flashed camera for comparison (it was a 2-pack) that's not even been powered on as yet.

roleoroleo, 2022-06-16:
Check this discussion:
https://github.com/roleoroleo/yi-hack-Allwinner-v2/discussions/445

smashley3000, 2022-06-16:
Thanks, not sure how I missed that! I'll attempt the version you posted there and update that discussion with my results.

closed 2022-06-16 by smashley3000